mrehkopf/sd2snes #53: No sound in MSU1 video player on higan 094 + fix ;-)

Ramsis-SNES, 2015-04-20:
I noticed that the MSU1 video player included in the sd2snes source package doesn't work correctly on higan 094 (there's no sound to be heard). This seems due to the fact that the code doesn't set the volume.

Adding these lines somewhere in the ``msu1init`` routine (e.g. after line 15) fixed it:

```
	lda #$FF
	sta MSU_VOLUME
```
We should be aware that this means there exists a discrepancy in MSU1 implementation between sd2snes and higan 094. ;-)

mrehkopf, 2015-04-20:
http://forums.nesdev.com/viewtopic.php?p=125863&sid=9d0f832d4cee24791dae503c8187089b#p125863
1st paragraph
Initial volume is kind of mushy in the standard. ;)

mrehkopf, 2015-04-20:
Nevertheless, I agreed upon initial volume 0 and the player will be fixed accordingly

Ramsis-SNES, 2015-04-20:
Cool, thank you! :-)

mrehkopf, 2015-06-22:
This change is now implemented on sd2snes and will be in the 0.1.7 release.

Ramsis-SNES, 2015-06-22:
Sounds good! :D

borti4938, 2015-09-01:
There are several MSU1 videos out there who are not setting the volume register. Hence, with the current implementation you hear nothing as the initial volume is 0.
This 'opposed issue' was reported to me by a snesfreaks-user. I undo commit 412127ed5f71848490c0ac73486f0348d44ae619 for him which resolves his 'problem'.
Maybe it could be useful to set the initial volume using the sd2snes-config

mrehkopf, 2015-09-02:
I'm sort of opposed to going the MS way of supporting all old cruft :D But I'll see to getting a fixed version of my player binary out in a timely manner. So far I know only of my code being affected.

mrehkopf, 2016-10-06:
"timely manner" he said 👍 💯 With the sample rate now fixed at 44100Hz the audio tracks would need to be resampled too! Oh, what to do...

closed 2018-11-20 by mrehkopf