mrehkopf/sd2snes #55: tetris attack, flickering in 1 player stage clear mode

mrbrianedwards, 2015-05-12:
n Tetris Attack, in 1 Player Stage Clear mode, there is a wild flickering while you play, it’s strange, ’cause some foreground elements stay normal, while the background and other parts flicker. Only happens in this 1 Player Stage Clear.

All other modes look fine.

It’s reproducible.

I don’t have a real Tetris Attack cart to compare.

Seems strange, this is the only game I’ve play on SD2SNES that has weird flicker issues.

I'm using this firmware:
Firmware v0.1.7 preview 3

mrehkopf, 2015-05-12:
Thanks for reporting. I'm quite sure it is caused by the IRQ hook routines for in-game buttons and cheats. I'm going to publish a less invasive version shortly which should improve it. Ultimately some games won't like the hook routines at all so I will make an option to disable them.
Meanwhile they can be disabled by pressing L+R+Start+Y at runtime (at least 10 seconds into the game).

crispycritter911, 2015-05-14:
It is because of the in game routines. I've been only able to play it by using an older firmware or disabling the in game routines.

mrehkopf, 2015-06-22:
Cannot be ultimately fixed as long as the in-game routines are in place. v0.1.7 will have an option to disable them by default.

mrehkopf, 2016-02-23:
Menu option to disable in-game hooks has been implemented in 10e9429 and 926673e.

closed 2016-02-23 by mrehkopf