mrehkopf/sd2snes #56: Shin Megami Tensei II.

ceramiclion, 2015-06-14:
Shin Megami Tensei II on SD2SNES flickers on the world map walking near the buildings. This issue happens whether the game is patched for english or a clean rom. The patching however seems to make the flicker get worst.
It does not seem to flicker on any emulator however.
Thank you.

mrehkopf, 2015-06-22:
while in-game can you try pushing L+R+Start+Y and see if the flicker goes away?

ramapcsx2, 2017-01-07:
If this ever was an issue, it appears to be fixed in 0.1.7.d :)

devinacker, 2017-01-07:
Probably was an issue with ingame hooks like @mrehkopf suggested, then.

mrehkopf, 2017-01-13:
Looks like I had completely forgotten about this issue. Good to hear that the latest hook improvements fixed it. :)

closed 2017-01-13 by mrehkopf