mrehkopf/sd2snes #57: Save state support

Skarsnik, 2015-06-20:
I could be nice to have a save state support.

Some people hacked game like Zelda 3 and super metroid to add save state support by copying the ram in the sram. (Using a second controller to load/save). Maybe sd2snes could have something similar to this :)

xadox-1st, 2015-08-21:
That would be a great feature.

hyrulebr, 2016-04-15:
Something like mega everdrive would be very nice!

ramapcsx2, 2016-06-01:
Is everyone is aware that on the SNES, a save state function can not include the APU state?
This will lead to sound being off or weird most of the time, and in some cases it will break the game.
Most of the time though, the ship will correct itself on a level load.
A savestate feature would be awesome, even with this limitation :)

gnadenlos, 2018-04-22:
usb2snes + savestate2snes work quite well, but they require an active USB connection to patch the games in memory every single time we want to use the feature.

A solution like that without the requirement of an USB connection would be a great start and much better than having no savestate support.

Even if you only integrate the existing code from redguy directly in the SD2SNES firmware and menu would be a nice improvement and give us quicksave / quickload (Select + R & Select + L) in a lot of games without the need for USB connections.

mrehkopf, 2021-08-02:
Savestates have now been integrated with the current firmware beta release (v1.11.0b1), with some compatibility improvements (and still room for improvement of course).

closed 2021-08-02 by mrehkopf