mrehkopf/sd2snes #59: Mario Kart run off course when it is not off course

RockinKat, 2015-07-02:
In Mario Kart, if you get too far off the course, a turtle in a flying cloud will hook you on a fishing line and return you to the center of the track. Playing the game on the sd2snes, the game activates this even if you're still within the track or just over the border. Comparing it to playing with the regular game cartridge, it is not working properly as when playing from the regular game cartridge it is no where near as sensitive letting you drive off-road along the track a lot more.

mrehkopf, 2015-07-02:
I've had rare occurrences of this behavior. v0.1.7 will have a reworked bus sampling which looks promising in development stage.

mrehkopf, 2015-08-20:
Please check if this has improved with v0.1.7pre4.

mrehkopf, 2016-02-23:
Very likely finally fixed in 688bd5d.

closed 2016-02-23 by mrehkopf

RockinKat, 2016-04-27:
Hi,

I had to pack up all my things due to a move and haven't had a lot of time to deal with other things...

I recently broke out my SNES and SD2SNES and I updated to v0.1.7b today and tried out Mario Kart and it's still doing it.

https://youtu.be/ltJY250CS_w

crash at 1:55: https://youtu.be/ltJY250CS_w?t=1m55s

off course at 2:38, followed by a crash later on same race. https://youtu.be/ltJY250CS_w?t=2m38s
https://youtu.be/ltJY250CS_w?t=2m58s

crash at 3:50 https://youtu.be/ltJY250CS_w?t=3m50s

off course on first corner repeated at 4:50 https://youtu.be/ltJY250CS_w?t=4m50s
followed by crash at 5:10 https://youtu.be/ltJY250CS_w?t=5m10s

mrehkopf, 2016-04-27:
Just to be sure, is your dsp1b.bin completely intact? ;)

RockinKat, 2016-04-28:
Yes, dsp1b.bin is present in the sd2snes folder.

This is the file copied from the SD card: http://www.thecowsaysmoo.org/rockinkat/files/dsp1b.bin

ramapcsx2, 2016-06-01:
From your video, it looks like you're running a PAL rom?
You can see a glitch on Toad in the intro. That happens when you run the wrong region of the game.

RockinKat, 2016-07-05:
Ah. PAL vs NTSC... I found a different rom set and the issues are gone. Derp. Thank you.