mrehkopf/sd2snes #60: Star Ocean: world music stops playing after a battle

orfjackal, 2015-07-25:
In Star Ocean, the world music plays normally when a game is started/loaded, but after a battle is finished there is no more world music - it's all silent (except for special effect sounds). The battle music still plays in battles after that, though.

This problem happens on SD2SNES firmware v0.1.7pre3. On v0.1.6 everything works fine.

The version of the game used was "Star Ocean (J) [T+Eng1.0_DeJap]" with the [no S-DD1/96Mbit hack](http://www.romhacking.net/hacks/614/).

mrehkopf, 2015-08-05:
Initial guess is that the in-game hooks are at fault. Does it work better when you push L+R+Start+Y on the world map (or somewhere in-game), then go into battle?

orfjackal, 2015-08-05:
You were right, pressing L+R+Start+Y before the first battle avoids the problem.

leonquest, 2015-10-07:
since when S-DD1 works on SD2Snes? the website says it hasn't been implemented yet.

DoctorDan1986, 2015-10-07:
There is an S-DD1-free patch that expands Star Ocean out to its
uncompressed size of 12MB.

leonquest, 2015-10-07:
Funny story, I answered my own question, as I headed to romhacking.net after posting it and discovered the hack. Thank you for your reply, good work on the new firmware!

ramapcsx2, 2015-10-22:
Note that this game is known to do this. Sound stopping after battles was (still is?) a problem on emulators as well.

mrehkopf, 2015-10-22:
Interesting! Regarding emulators, does this apply only to the 96Mbit no-SDD1 patch or to the vanilla ROM as well?

ramapcsx2, 2015-10-29:
I (play)tested it ages ago. Back then we used graphics decompression packs and far less mature emulators ;p

mrehkopf, 2016-06-20:
Maybe not a hook issue but a 96Mbit hack issue then. ;)

closed 2016-06-20 by mrehkopf