mrehkopf/sd2snes #61: SSF2 glitchy sprites

dasi, 2015-08-04:
Hello. I'm getting jerky, glitchy character graphics on SSF2. I'm using a 1CHIP-03 Super Famicom with SD2SNES #123 (which I bought back in 2012 but have only just got around to trying out). The same thing happens in both firmware 1.6 and 1.7 preview 3. I've checked and the ROM is good. Strangely, sometimes if I load another game first and then load SSF2 after, it'll run normally without the glitching. I haven't seen anything similar occur in the dozen or so other games I've tried.

[image]
[image]

mrehkopf, 2015-08-05:
Oh dear, that sure looks like the SNES DMA engine is acting up again. Nawww.
I still don't know why exactly but sometimes the SNES just doesn't carry out DMA transfers (CPU is halted for a few cycles after DMA is triggered, then just continues normal execution as if nothing happened).
This seemed to happen when initializing DMA registers without using them, then pulling the /RESET line to reset the console.(there was severe sprite corruption in Super Ghouls N Ghosts because of that - you might want to check on it, it will occur as soon as level 1 starts.). Using HDMA together with general purpose DMA seems to play a role as well. I'm not sure how I can solve this, the sd2snes menu itself makes heavy use of (H)DMA so hm. Guess I'll try and reproduce it at first.

dasi, 2015-08-05:
)-: Yes, I see the same thing in Super Ghouls'n Ghosts and SF2 Turbo as well. If I can help with testing in any way let me know.

dasi, 2015-08-14:
Hmm. There's sometimes also glitching of the file name text as it appears on screen when using the d-pad to scroll through or switch pages in the sd2snes menu.

mrehkopf, 2015-08-15:
Yes, that is part of the same issue. DMA updates just don't work sometimes. I kind of hate to ask but did you double-check that the firmware upgrade actually worked? Firmware version is displayed in the "System Information" screen, accessible via menu (X button). This should really have been fixed since v0.1.4a.

dasi, 2015-08-15:
Hi mrehkopf . Yes, "Firmware version: 0.1.6".

I bought another Super Famicom last week, a 1CHIP-02 this time, and after some testing the 1CHIP-02 doesn't appear to suffer from the glitching/DMA issue. Another thing I noticed was that the 1CHIP-02 doesn't suffer from the RF-like interference pattern that I see on my 1CHIP-03, either. As presumably you haven't had any other reports of this issue since it was fixed in 0.1.4a, perhaps it's just this particular 1CHIP-03 Super Famicom that's a bit finicky, but then it appears to work fine otherwise. What I ought to do is test it with an actual SSF2 cart and see if it still occurs.

DoctorDan1986, 2015-08-15:
I'm wondering if there's an actual hardware issue with your particular SNES.  Just as an idle curiosity, have you run the SNES test cartridge ROM on that system, and seen if all the checks pass?  I've come across a few SNESes that have failed the APU, and one that had bad work RAM.

dasi, 2015-08-16:
Hi Doc. I've run the SNES Test Cart Ver1.03 burn-in test on my 1CHIP-03 for an hour and everything passes.

mrehkopf, 2015-08-31:
I moved some DMA stuff around and reverted it to "after the 0.1.4a fix, before cleaning up":
https://sd2snes.de/files/v0.1.7pre4-dma-201508311325/menu.bin
Can you overwrite your menu.bin with this and retest?

mrehkopf, 2015-09-03:
Another user who just reported the same problem with a Jr. reported that it has been fixed with the menu.bin posted above. So I'm closing this issue for the time being, please re-test when you get around to it :)

dasi, 2015-09-03:
Hi mrehkopf. I haven't been able to reproduce the problem when using the new menu.bin. Nice work.

mrehkopf, 2015-10-02:
I needed to change the DMA init a bit and would like to make sure it doesn't break this issue again.
Could you re-test for sprite corruption with this menu.bin?
https://sd2snes.de/files/v0.1.7pre4-dma-201510021116/menu.bin
That would be great :)

dasi, 2015-10-10:
Hi mrehkopf. Sorry for the delay. I haven't seen any glitching while testing with the menu.bin above.

mrehkopf, 2015-10-21:
Thanks :) 
Hopefully finally fixed in a81dffd.

closed 2015-10-21 by mrehkopf