mrehkopf/sd2snes #62: cx4 register file error / glitchy megaman x2

hippie131, 2015-08-19:
Im using a 1chip-01 north american snes. When I run Megaman x2 on my sd2snes rev F i get glitchy sprites as well as a glitchy title screen and the health bar as well. I also get the register file error when I load the game. On 0.1.6 i still got the error but the game played fine but on 0.1.7 preview 4 i get these glitchy graphics. I got my sd2snes from stoneagegamer almost a year ago.

mrehkopf, 2015-08-20:
I added some multi-cycle constraints to the Cx4 core so it would meet timing. I guess one or more of them are still hazardous.

mrehkopf, 2015-08-20:
Can you replace sd2snes/fpga_cx4.bit with this file and see if it works better?
https://sd2snes.de/files/v0.1.7pre4-cx4-201508200644/fpga_cx4.bit
I tightened a constraint regarding transfer to the register file.

hippie131, 2015-08-20:
Just tried the replacement fpga_cx4.bit file and it fixed the register file error but now I get an immediate rom error (tested on multiple cx4 roms) and the graphics are even more glitchy

mrehkopf, 2015-08-20:
Bugger :) OK, I'll try something else...

hippie131, 2015-08-20:
ah i got the revision wrong. Sorry, it is rev E1 but that is just the sticker on  the back of the cartrage stoneagegamer put on. does that matter at all? is there an easy way to determine this info from the GUI or would I need to open it?

hippie131, 2015-08-20:
Just did more research. I see rev F fixed some 1chip issues. Could this be the problem?

mrehkopf, 2015-08-21:
Most likely not, it was a fix for some analog noise but otherwise shouldn't have any impact.

mrehkopf, 2015-08-21:
https://sd2snes.de/files/v0.1.7pre4-cx4-201508210642/fpga_cx4.bit
Try again please? ;)

hippie131, 2015-08-21:
that fixes the glitchy graphics on both x2 and x3. Thanks. I still get the register file error on both and the immediate rom error on x3 only.

mrehkopf, 2015-08-21:
Hm, better than nothing I guess... going to look further.

mrehkopf, 2015-08-21:
What SNES master clock frequency is shown on the "System Information" screen?

hippie131, 2015-08-21:
21477746Hz

mrehkopf, 2015-08-21:
Good, so it's not like the FPGA is running faster than spec (due to detuned quartz or something).
Still trying to achieve timing closure...

mrehkopf, 2015-08-22:
https://sd2snes.de/files/v0.1.7pre4-cx4-201508221116/fpga_cx4.bit
This one meets timing with some margin, without any multicycle constraints in the Cx4 core. If this doesn't work properly something else must be wrong =)

hippie131, 2015-08-22:
That fixed the immediate rom error on x3 still no glitchy graphics. Both x2 and x3 still have the register file error. Thanks for all the help. The register file error was there in 0.1.6 so do you think its fixable?

mrehkopf, 2015-08-23:
Honestly I don't know at this point :) I need to determine possible causes for the error but I cannot reproduce it on any of my sd2sneses or SNESes. So it might take a while. Do the games work perfectly otherwise?

hippie131, 2015-08-23:
They do seem to

Ripthorn, 2015-08-23:
SNES 1CHIP-03 and SD2SNES Rev E here.
No issues for me on both games with the preview 4.
Faulty SD2SNES maybe?

hippie131, 2015-08-23:
Could be but i cant dish out another 300 bucks to get another. Dont think i can get an exchange either since ive had it over a year. Plus everything seems to work fine. Just that error. Game plays fine.

hippie131, 2015-08-23:
Also the first replacment fpga file fixed the register file error but the glitchy stuff remained

mrehkopf, 2015-08-24:
Do you get a "REGISTER FILE ERROR" on every boot or just when you run the full Cx4 test (hold B on controller 2)? I think the game only runs the IMMEDIATE REGISTER and IMMEDIATE ROM tests otherwise but I'm not 100% sure...

hippie131, 2015-08-24:
Every time i boot a cx4 game

mrehkopf, 2015-08-26:
https://sd2snes.de/files/sd2snes_firmware_v0.1.7pre4cx4_1.zip
I changed more stuff around so multiple files are affected this time... please unpack this like installing a regular firmware update.

mrehkopf, 2015-08-26:
(please verify version "0.1.7 preview 4cx4_1" in system information)

hippie131, 2015-08-26:
that fixes megaman x2 and x3. they both work perfectly without errors. I noticed that obc1 was also altered so i tested metal combat too. it seems there are no sprites. I can hear the music and see the background moving and can hear and see myself getting damaged but no enemies.

mrehkopf, 2015-08-26:
Well, that's good to hear about Cx4 at least :)
Hm. Can you confirm that fpga_obc1.bit is present and intact on card (some 70kB) and that OBC1 works with "vanilla" v0.1.7pre4?

hippie131, 2015-08-26:
confirmed fpga_obc1.bit is indeed on the card at 71kb. obc1 does work with the vanilla v0.1.7pre4.

mrehkopf, 2015-08-27:
I see... maybe I should use the amended data line sampling on the other fpga files too.
Does Mario Kart work properly? Or are the drivers disappearing, jumping around, etc.?

hippie131, 2015-08-27:
Whoa its working fine now. Obc1 works cx4 works perfectly. Everything is good now. Mustve not copied properly. Sorry and thanks for all your help and work on this.

closed 2015-08-27 by hippie131