mrehkopf/sd2snes #64: Intermittent SPC playback problems

BMF54123, 2015-10-08:
After upgrading to 0.1.7 pre4 recently, I've experienced some intermittent issues when playing SPCs with the built-in player that did not exist in 0.1.6. I've observed notes playing at the wrong pitch and samples looping improperly (e.g. a drum sample loops when it shouldn't, causing a machine gun-like effect), primarily in the U.N. Squadron set. Backing out and reloading the SPC (or loading another one) usually fixes the problem, though I believe there was one instance where switching to a different game's set caused the issue to reoccur. This was observed on both an SNS-CPU-RGB-02 and an SNS-CPU-GPM-?? (don't recall the number, and the console is currently in use so I can't check). I haven't experienced any problems loading actual games.

Could this be related to the uninitialized APU RAM issue that was fixed recently?

mrehkopf, 2015-10-08:
The APU init commit is ahead of v0.1.7pre4 so it must be something else... I'll check for any potentially harmful changes.

mrehkopf, 2015-10-08:
APU RAM init might fix the issue though, or at least make it more reproducible. Can you test with this menu.bin? https://sd2snes.de/files/v0.1.7pre4-dma-201510021116/menu.bin

mrehkopf, 2016-10-03:
Figured it out most likely, it was a bug in the bulk RAM transfer routine in the SPC player code. It didn't account for the SNES CPU getting stalled by HDMA and has been present ever since the SPC player has been around ;)

mrehkopf, 2016-10-03:
5335990 addresses this issue and hopefully fixes it.

mrehkopf, 2016-10-06:
Integrated in v0.1.7d

closed 2016-10-06 by mrehkopf