mrehkopf/sd2snes #65: Saves folder for .SRM Files

OverloadedRAM, 2015-11-11:
I was wondering if it would be possible to make the .srm files save to and load from a specific folder instead of the same location of the rom.

xadox-1st, 2015-11-12:
I would also like such an option.

Ramsis-SNES, 2015-11-12:
Seconded. :-)

mrehkopf, 2015-11-12:
Can we implement this statically or do we need it as an option?

MatthewCallis, 2015-11-12:
Sounds like a breaking change for anyone already using saves if you went with it statically. It would leave a much cleaner directory structure though, :+1: statically.

magu, 2015-11-12:
> Sounds like a breaking change for anyone already using saves if you went with it statically. It would leave a much cleaner directory structure though,  statically.

Indeed statically!

OverloadedRAM, 2015-11-12:
Statically.

Ramsis-SNES, 2015-11-12:
Make it an option. (Don't get me wrong -- I'd prefer it implemented statically myself, but you'd definitely set the cat among the pigeons if you did. ;-) )

borti4938, 2015-11-13:
Nice idea :) 
+1 for a static

Maybe a useful and very userfriendly implementation could be something like:
* look for *filename*.srm in /sd2snes/srm/
  * if present:
    * load it
    * end here
  * if not present:
    * look in folder of loaded game file for *filename*.srm
      * if present:
        * copy it to /sd2snes/srm/
        * delete old file
        * load the new one
        * end here
      * if not present:
        * create new *filename*.srm in /sd2snes/srm/

Quite a long checking chain, but the user don't have to copy all of his save games spreaded over the entiere SD card to a single folder. He don't have to care about this change and proceed as always with a firmware update.

mrehkopf, 2015-11-13:
Get out of my head :D

borti4938, 2015-11-13:
:grin: :sweat_smile:

xadox-1st, 2015-11-13:
An automatic move function would be great, but for start I would already be happy with a non automatic static srm folder.

borti4938, 2015-11-27:
My trail: [Downloadlink] (https://dl.dropboxusercontent.com/u/18680455/SNES-Projects/sd2snes_nightly/sd2snes_v017pre4.3.zip)
If somebody wants to test it, go ahead. *Please forward any bugs regarding the common srm-folder to me.*
It's a current build of my fork with some minor changes as discussed here. (just seven lines of new code: new SRM-Files are created in /sd2snes/saves/ and if not present old ones are tried to move)

xadox-1st, 2015-12-14:
Tested with around 10-20 games. Working fine for me.

Ramsis-SNES, 2016-01-04:
@borti4938, I've only tried your build with a couple games, but it seems to work as expected. :D

I wonder though what happens if there are multiple ROMs with the same name spread over several directories, e.g. like this:

\mama\zelda3.sfc
\papa\zelda3.sfc
\kevin tigris\zelda3.sfc
\vanessa lavinia\zelda3.sfc

In case everyone shares the SRM that was loaded last, I could imagine a domestic occurring rather soon. :D

mrehkopf, 2016-01-04:
An alternative would be to recreate the full ROM path under the SRM 
folder. But would that really still be an advantage over putting the 
SRMs with the ROMs?

borti4938, 2016-02-12:
Reproducing the folder structure within the /sdsnes/saves/ folder would be nonsense in my point of view.
Actually I do like the option to share SRM-files with my implementation. Now I do have multiple copies of some games in different folders because I can fit it in multiple categories (my folder structure).
If a family uses the same sd2snes with the same SD card (how often does that appear???) and if they don't want to share the SRM file, they do have the option to slightly rename their game copies.

mrehkopf, 2016-02-15:
Requesting pull request :D

mrehkopf, 2016-06-20:
Implemented in v0.1.7b. (with bug #77)

closed 2016-06-20 by mrehkopf