mrehkopf/sd2snes #68: Freeze when user attempts to load a game from the "Recent Games" list that was deleted or moved

Ramsis-SNES, 2016-02-11:
Title says it all, the system just freezes on the "Loading" message. Tapping Reset will mess up the menu, a real power-cycle is needed.

One solution would be to just display a "File not found" error message. Or even more convenient, have no longer valid entries simply vanish from the list. :)

borti4938, 2016-02-11:
you are right - good find on that. I would prefer a validity check after each new power cycle before the list is dumped for the SNES. Not that hard to implement. I gave that one a try ;)

mrehkopf, 2016-02-11:
How slow is it? ;)

borti4938, 2016-02-11:
I don't feel a difference between with and without that check even with a full list.

borti4938, 2016-02-11:
I see: one could change the running order in the middle loop: https://github.com/borti4938/sd2snes/commit/1150572944bf3b79eae54277772b23506735929d#diff-a3b2a35e1d5756a4ca4b7049d18e630eR146
Might be confusing if someone think about why to run backwards there. It was from my first idea, where I wanted to reuse the cfg_add_last_game()-function.
Should be no differnce though.

Ramsis-SNES, 2016-02-11:
I hardly understand a word that's been said, but I'm glad this is being taken care of, anyway. :D

mrehkopf, 2016-02-15:
Requesting pull request ;)

mrehkopf, 2016-03-14:
There might be an off-by-one in there :D 
`file_open (/demos/elix-smashit-pal.sfc, 01) = 0
[...]
file_open (/msu1/[krom]/Touhou-BadApple!30FPS4ColInterlace/Touhou-BadApple!30FPS4ColInterlace.sfc, 01) = 0
file_open (/msu1/sndtest/msu1snd.sfc, 01) = 0
file_open («Ý¿ï‚Wp]µÙÝ;L¤þznÿ¨!)j_óÿe«Md_ëýû¨%B§Í{•äð`Þ+ÿO‚@îbîŸxdˆ„Ÿ¿}¹, 01) = 6`

borti4938, 2016-03-21:
I had something like that implemented incl. a backup-file creation (just in case). But I never tested it so far. I'll try to do that this evening.

closed 2016-06-20 by mrehkopf