mrehkopf/sd2snes #70: MSU-1 video cracks up

borti4938, 2016-02-21:
As titled: MSU-1 video flickers and cracks up. Some frames are fine, most frames are corrupted. This is probably related to the MSU-1 resume feature (SHA 453a36aa651e667ff0b19bd86f1092749e52dacb) All frames are fine if I undo the implementation.

Here is a pictrue which shows what I mean: [uploaded picture](https://dl.dropboxusercontent.com/u/18680455/SNES-Projects/sd2snes_nightly/msu1-video.JPG)

I've tested this with Super Road Blaster and two SD cards:
- 1st SD card has access times: 0.548ms avg and 0.885ms max
- 2nd SD card has access times: 0.319 ms avg and 3.480ms max

mrehkopf, 2016-02-22:
Oh that. That's the new Psychedelic Mode, haven't you heard of it? :smirk:

mrehkopf, 2016-02-22:
Found it :) In verilog/sd2snes/msu.v the status has been extended to 8 bits but the status_out bus is still 7 bits.

borti4938, 2016-02-22:
You are right - the Psychedelic Mode is gone :grin:

mrehkopf, 2016-02-22:
Fixed in d281b01

closed 2016-02-23 by mrehkopf