mrehkopf/sd2snes #77: Firmware 0.1.7b doesn't read or write game saves

redeemer665, 2016-06-11:
Fw0.1.6 works normally with my 4GB card with no bad blocks or errors, reading and writing saves like it should.
After fresh upgrade to 0.1.7b by deleting old files first, only the red light is flashing intermittently when playing any game.
Starts by flashing about 8 times first, then 7 times, then 6, maybe 5 and half. Doesn't seem to be any clear error code to me.

How can I debug this further?

redeemer665, 2016-06-12:
Tested another card with the same contents and so is the result same as well.

mrehkopf, 2016-06-12:
Did you unpack the empty sd2snes/saves folder from the zip? it must exist in order for saved games to work.

redeemer665, 2016-06-19:
... Yes, that was the reason. Thank you and sorry for the trouble, I was using an old version of Winzip which doesn't see (?) empty folders inside the zip. I even failed to find enough information about this from your blog!

magu, 2016-06-20:
Even though this is closed now, I suggest a feature where the firmware checks to see if the saves folder exists, and, if not, creates it. Makes for a better and more seamless user experience. At least some sort of error should exist, and then you might as well just fix the issue instead.

devinacker, 2016-06-20:
I agree, it's definitely more convenient when writing files to a directory for the directory to be automatically created if it doesn't exist, rather than assuming that the user will know it's expected/required to be there beforehand.

mrehkopf, 2016-06-20:
I obviously agree ;) I had to analyze the cause myself at first. This was a contribution (#65) and I failed to review it thoroughly. My bad.
Reopening for tracking.

mrehkopf, 2016-06-24:
resolved in 03835b3

mrehkopf, 2016-08-22:
integrated in v0.1.7c

closed 2016-08-22 by mrehkopf