mrehkopf/sd2snes #78: NES backward-compatibility

eadmaster, 2016-07-18:
A [software solution for this was discussed some years ago in this forum](http://forums.nesdev.com/viewtopic.php?t=5826).

Btw recently many NES implementations in FPGA come up, so it may possible to reuse some of their code and add NES backward-compatibility by emulating the missing chips in hardware?

mrehkopf, 2016-07-23:
I wouldn't say this is generally impossible, though it might be impossible to reach 100% compatibility. Using the 65816 emulation mode should work in most cases unless a game uses 6502 illegal opcodes.
APU can be emulated by the FPGA and output directly via the sd2snes DAC (normally used for MSU1 audio) so that is out of the way.
The trickiest part is getting any picture out of the system. When using the FPGA to emulate a NES PPU and render the picture, the SNES data bus bandwidth becomes the bottleneck. It is impossible to stream a NES resolution picture to VRAM at 60fps.
Alternatively the FPGA could provide a sort of compatibility layer, acting as a PPU to NES code but translating the PPU state so it becomes usable by the S-PPU. In this case it would be necessary to periodically interrupt the CPU to update sprite tables, possible preload some HDMA tables for the upcoming frame, map the rest of the registers accordingly etc. I do not know enough about the NES architecture to answer this thoroughly right now, though ;)

ramapcsx2, 2016-07-23:
Hmm, didn't SuperFX games render on the cartridge and the SNES just pulled the picture in intervals?

mrehkopf, 2016-07-23:
Yes, and so does the Super Gameboy. But both use either lower number of colors, lower resolution, lower framerate, or any combination of the three ;)

mrehkopf, 2016-07-23:
CHR bank switching will probably kill the idea (there are games with more CHR ROM than the SNES has VRAM).

eadmaster, 2016-07-26:
Having 100% compabilibility is not really important for me.
if i get it right, it would be possible to suuport only games having CHR<=128kb in [this list](http://tuxnes.sourceforge.net/nesmapper.txt).
Alternatively, can't you store CHR in the FPGA?

juef17, 2021-08-02:
So this is not exactly a sd2snes feature, but still related to this request: https://github.com/Myself086/Project-Nested