mrehkopf/sd2snes #80: Fix saving for MSU games.

DieKatzchen, 2016-07-23:
I've noticed that all the MSU enhanced games I've seen have required you to reset the system to write the saves to disk. I assume this is because the MSU is using save ram as work ram. Is it possible to add the ability to force a write via some signal the game modder can add to the save code? It could also be used to patch in support for original games which require this workaround.

mrehkopf, 2016-07-23:
The reason it can't save is that the MSU1 system claims the SD Card at all times. (Especially for the MSU data extension it is important that a game can be served data at any time because the MSU1 standard has no flow control mechanism).
I'm not opposed to a feature like this at all but I doubt byuu will allow something like it into the MSU1 standard (which he de-facto "owns")...

DoctorDan1986, 2016-07-24:
I’m going to assume that saving works more or less the same in all games.  So why not have it set up so that when saving occurs, MSU-1 functionality is suspended or stopped until the save is complete, whereupon it is restored?  Yeah, the best that we can hope for is to restart the MSU-1 track, but it’s still a thought.

qwertymodo, 2016-08-05:
Would it be possible to wait until bits 6 and 7 of the status register were clear, set both of those bits to mark the MSU-1 as busy, write the save, then clear those flags, and allow the MSU-1 to continue?  Granted, this would basically mean you could only save between track changes, which isn't ideal, but it's a thought.  Or, I believe you currently have the "last game loaded" saved in a config file, would it be possible to use that to have it save the battery-backed SRAM out to the last loaded game file upon power up?  Or is the CMOS battery only used for the RTC and not for the SRAM?  Just a few ideas...

mrehkopf, 2016-09-12:
The secondary RAM on the sd2snes is battery backed but currently unused =) It is intended to serve exactly the purpose you describe.

The other method can be unreliable. Occasional audio dropouts aren't that much of a problem, it's the data extension where things get ugly.
Once a data offset has been requested and the busy flag is cleared, it can never be set again because the MSU1 is designed that way - the SNES program will NEVER check again if data is ready for reading, it just assumes it can go on forever at full DMA speed. This means that if data has been requested even once it will hog the card forever. If we save inbetween and the game decides to continue streaming data just then, it will run into a buffer underrun, reading old buffer contents repeatedly. Depending on the kind of data this might result in anything from an intermittent graphical glitch to a full crash.

I can think of some heuristics however. If the MSU file size is 0 bytes the data extension will never be used. If a game only uses the data extension to play an opening FMV and nothing else, the file read pointer will be at the end of the MSU file when it's done. These are cases when we can assume that the data extension is not going to be used subsequently.
Super Road Blaster frequently requests new data offsets for each corresponding video clip so that can actually work "by design".

qwertymodo, 2018-03-22:
Would it at least be possible to commit the SRAM to the card on a short reset, and not just the long reset that takes you back to the menu?  I've lost more than one save due to forgetting that it has to be the long reset, so I short reset, confirm that the save is there, and power off, only to discover that it never actually wrote the save to the card.

mrehkopf, 2019-06-18:
Surprised I didn't reply to this yet... you're making valid points. I fear Super Road Blaster will have to increase its MSU1 failure timeout from 100ms to something longer ;)
I'll need to make a checklist with all the occasions to sneak in a save.

codemann8, 2020-01-02:
I noticed a couple commits were made around this issue.  Is this close to release-ready?

qwertymodo, 2020-03-14:
I had a couple of thoughts about this today. If the .msu file is 0 bytes, then you can commit the save file on every write to $2005 while the audio busy flag is high, and if no track is currently playing, you can commit it on any write to $2003 while the data busy flag is high. Now there's a chance that a track has been loaded but not started, or else paused when the second case occurs, but that can be mitigated by just latching any writes to $2007 during the save commit and executing them as soon as it's done.

mrehkopf, 2023-03-07:
I did something to that effect but now some games don't like it :) Super Road Blaster actually fails with an MSU1 timeout; Also #155. Appears to need some polish still.