mrehkopf/sd2snes #81: Issues when backing out of the SPC player

SwordlessWarrior, 2016-07-25:
I noticed a little problem that happens when you select a SPC file when a file with a name that's too big to fit on the screen is scrolling:

This is how it looks before it starts scrolling:
[image]

and this is how it looks when I back out after it scrolls:
[image]

But when I scroll down the list or go to another folder it fixes itself.

mrehkopf, 2016-07-25:
Heh, it scrolls all file names because the scroll displacement isn't reset before going back to drawing the file list.

mrehkopf, 2016-09-12:
Where did you get the SPC packages with the nice file names? I can only find "thg-1-01.spc", "thg-1-02.spc" etc.

mrehkopf, 2016-09-12:
Nevermind, found nice archives at http://spc.joshw.info/ ;)

mrehkopf, 2016-09-13:
Fixed in c85d302

mrehkopf, 2016-10-06:
Integrated in v0.1.7d

closed 2016-10-06 by mrehkopf