mrehkopf/sd2snes #82: Micron SRAM Obsolete

ccoulton, 2016-09-30:
I have a Digikey cart I can share of the whole SMT components, however, there is a small hitch.  The SRAM is no seems to be unavailable.  I can share this for anyone in the US for a build, however i'm not sure if any 8x10 pin ram should work, or if there is a reason for that chip directly.

mrehkopf, 2016-10-03:
Unfortunately there is no drop-in replacement for the PSRAM part used. Sorry about that!
ISSI started making PSRAM chips but they are only half the capacity so two would have to be used, requiring new PCB layout and FPGA code.
Alternatively SDRAM could be used but that too would require a new PCB design and significant changes to the FPGA cores, with increased resource requirements.
That said, chip brokers (and alibaba dealers) still seem to have the part available.

closed 2016-10-03 by mrehkopf

ccoulton, 2016-10-04:
found it on alibaba as well soon after... I need to decide if i'm jump off and try and hand solder this project... or learn how to dyi reflow.