mrehkopf/sd2snes #83: 0.1.7.d background corruption in Justice League Task Force

rimaille, 2016-11-29:
Hi Ikari, (thanks for your hard and amazing work, really appreciate my sd2snes!)

Using 0.1.7d, i noticed a corruption of background in Justice League Task Force , this issue always occure after a full combat, and return on player selection screen. Sometimes this affects character selection screen, sometimes it is battle background that is flickering.
After choosing character / finishing the round, everything turn back OK. This is not really a problem, the game is  still very playable.

(comments are closed on your site, that's why i post here instead of in game hook compatibility page)

respects,
ekyo

mrehkopf, 2016-11-29:
Thanks for reporting, I'll check and add it to the list!
On a side note, comments aren't closed generally. But the spam filter seems eager to sieve out old user agents, more specifically IE8 on Windows XP. Are you using that? If not, please let me know.

rimaille, 2016-11-30:
You probably found the problem : my user agent is "Mozilla/5.0 (X11; Linux x86_64; rv:50.0) Gecko/20100101 Firefox/50.0" maybe this is too new or too anecdotal for the spam plugin :)
I'll try with a user agent changer plugin to find if it's OK with that (when i'll have something to say)

mrehkopf, 2016-12-03:
Would you mind posting another comment? Anything will do. I enabled temporary logging of blocked comments so I can get a reason code for your comment being blocked.

rimaille, 2016-12-04:
Just did, 2015/12/04 17h57 on the https://sd2snes.de/blog/compatibility/in-game-hook-compatibility#respond page.
Got "ERROR: Comments have been temporarily disabled to prevent spam. Please try again later."
My only addon uBlock Origin was disabled.

rimaille, 2016-12-04:
I changed my referer to send a chrome / windows string, same result.

mrehkopf, 2016-12-04:
Got it:
```
Failed Test Codes:      ['REVD4110']
```
Plugin source:
```
        $banned_tor_exit_nodes = array(
                /* Banned Tor Exit Nodes */
[...]
        "REVD4110" => "~\.proxad\.net$~",
[...]
```
Looks like the spam filter don't like proxad one bit. 😅 IMHO it's a bit far-fetched to consider all proxad.net IPs Tor exit nodes but they seem to be notorious for spam or worse.

rimaille, 2016-12-04:
Thanks for watching your spam plugin traces.
Proxad, is the 2nd internet provider in France (via Free), and they also provide rented servers (via online.net).
I agree with you, it's a bit far fetched =)

mrehkopf, 2016-12-05:
Can you try posting again? I disabled that specific filter.

rimaille, 2016-12-05:
Thanks Ikari, i left a comment successfully (no error message), but as coments are moderated, i can't see yet if it worked.

closed 2018-11-20 by mrehkopf