mrehkopf/sd2snes #85: Autoboot feature for a pre-selected ROM

mrehkopf, 2017-03-22:
Implement a feature to have the sd2snes boot a pre-selected ROM automatically on boot, without user interaction and without showing the menu.
Possible scenarios:
- Interactively select a file for subsequent auto-booting with a bail-out button combination to go back to the menu (more complex solution)
- Just expect a fixed filename (e.g. `autoboot.sfc`) and always boot that file if it exists (simple solution)

MatthewCallis, 2017-03-22:
The second / simple solution leaves the user with no possible way to get out of that game without access to a machine that can read and manipulate the filesystem to change the name of the `autoboot.sfc` or remove it.

mrehkopf, 2017-03-22:
To be fair they'd also need access to such a machine to put the file there in the first place though.

MatthewCallis, 2017-03-22:
Sure, but going to an Expo or something where you want it to autoboot then its over and you just want free play you're stuck.

magu, 2017-03-22:
With the bailout key combination, do you mean escaping from mid game out to menu or sequence to press while booting?

Simplest solution I can figure out would be just turning the SNES on with the A—or even just any—button pressed to clear the auto boot setting and revert to menu. Doesn't have to be any more sophisticated than that.

Optiroc, 2017-03-23:
This is actually a feature I would want to use at home since I usually play just one title until I'm done with it, and skipping going through the menu at each "cold start" would be a win.

Here's my simple suggestion: Have a toggle in settings called "autoboot into latest file" (or preferably something more succinct! "autoboot", "kiosk mode", "kids mode"...). When that is set SD2SNES will always automatically load the last file except when holding a certain button combination. Preferably something easy enough to press using only the right hand so you simultaneously can press `reset` without using your foot. `R + X` or possibly `R + X + A` are my suggestions.

ramapcsx2, 2017-03-28:
Simple solution is the autoboot.sfc + force enabled ingame key hooks + a memory resident flag that an autoboot game ran once. A user that enabled the feature is surely able to remember the back to menu key combo.
May I also suggest mapping the start button to acknowledge, so that starting the last loaded game is a simple double tap on start? This works great in Everdrives :)

DieKatzchen, 2017-08-06:
I dunno, I like Magu's suggestion best. Forcing ingame hooks obviously prevents using this feature with any game that has trouble with ingame hooks. Nintendo has a long tradition of holding a button while booting, though.

rarn, 2017-08-10:
Someone using autoboot will not go back to the menu often. In fact going back often means changing rom, which contradicts the need of autoboot. So I'd say there's no need for fancy bailout thing.
If the long reset could ignore the autoboot flag that would be the easiest, I think.

dhincardoso, 2017-09-25:
Man! Implement the simple solution, it's enough for now for most of us, leave the complicated stuff for later, better have an option, even a simple one, than none. I absolutely agree with @rarn

DhinCardoso, 2017-09-25:
Man! Implement the simple solution, it's enough for now for most of us, leave the complicated stuff for later, better have an option, even a simple one, than none. I absolutely agree with @rarn

DhinCardoso, 2017-10-11:
@tandraschko why you referenced this request to the #86 and then referenced #86 back to #85? =/

2dcomplex, 2020-07-13:
It looks like it's been a couple years since any activity has been posted. Just wanted to chime in and say this would still be a great feature and am in full support of it getting implemented if at all possible.

DhinCardoso, 2020-07-14:
> It looks like it's been a couple years since any activity has been posted. Just wanted to chime in and say this would still be a great feature and am in full support of it getting implemented if at all possible.

Agreed!

CianMclennan, 2023-12-10:
Just to keep this alive. I would also like this feature.

masterquest91, 2024-07-09:
2024 checkpoint. I would use this feature.

AkimSALADIN, 2025-03-22:
Hello! I need this feature, please !

@mrehkopf, can you contact me via Twitter or Discord? I need your help creating this feature. 
Just give me the path to and I make the changes.

Basically, load "autoboot.sfc" located in the root of the storage and boot it directly without going through the SD2SNES menu.

Thank you !

Ramsis-SNES, 2025-03-23:
> The second / simple solution leaves the user with no possible way to get out of that game without access to a machine that can read and manipulate the filesystem to change the name of the `autoboot.sfc` or remove it.

8 years later, I strongly agree with @MatthewCallis. :-)

AkimSALADIN, 2025-03-23:
Yes, I know, but I need this functionality for personal purposes, so I would like to know where (in the source code) should go to add this feature please ?

Btw, It's just a suggestion but to avoid being stuck with the "autoboot.sfc" file, we can imagine doing a manipulation that allows you to skip the automatic loading of "autoboot.sfc", for example, start the console and hold the R button on SLOT1 (or SLOT2).
This would allow you to load the SD2SNES menu and then disable the feature in the menu.

mektrix, 2025-06-19:
This would be useful for the Burn-in Test diagnostics rom, I'd love to use my sd2snes for this so I don't have to buy another cart.