mrehkopf/sd2snes #88: LED Brightness

kylophone, 2018-02-11:
The LEDs can be overbearingly bright during use. It'd be nice if there was a configuration to either control the brightness of the LEDs, or to turn them off completely.

mrehkopf, 2018-07-17:
Conveniently the firmware can already do this. Now I just need to expose it via a configuration option.

kylophone, 2018-07-18:
Awesome. This is a simple little feature, and it would be much appreciated!

ramapcsx2, 2018-07-18:
In some cases (cart shells I mean :p), the LED is not exposed and has to shine through the plastic.
If I wanted to make it brighter, I realize I would have to change the current limiting resistor.
Is there some value on how low one can go for these? 

@kylophone If the LED is too bright on a product, may I suggest the good old CD marker? :)

mrehkopf, 2018-07-18:
I am not sure what type of LED krikzz is using, I think he recently switched to a very bright type with no diffusion (clear body). 
Resistors are 680 Ohms by default, I had originally used low power LEDs @2mA each.
You could also thin the cart shell a bit from the inside =)

mrehkopf, 2018-07-18:
Seems that fairly bright LEDs (900-2800mcd) have a forward voltage of around 1.85-2V at 20mA, so if you calculate your resistor to drop 1.5V (3.3V-1.5V=1.8V) you should be fine. So the smallest "safe" resistor should be around 1.5V/.02A = 75 Ohms.
However, at the full 20mA the LEDs will contribute significantly to the total power consumption of the sd2snes (~100-120mA).

ramapcsx2, 2018-07-18:
Thanks!
I appreciate the formula and ballpark figure for the resistor :)
The shell has already been thinned out and it only helped a little. The LED can be seen in low light conditions now.
I'll try to get the LED current to maybe 5 to 10mA and see if that's enough for daylight visibility.
By the way, I'm only interested in the save / load LED. It's nice to see at what point a game actually writes out a save game :)

kylophone, 2018-08-14:
@mrehkopf, I can take a crack at this myself. Is there a place in the code I should be looking at?

closed 2018-12-18 by mrehkopf