mrehkopf/sd2snes #91: MSU-1 seek to current offset cache behavior

qwertymodo, 2018-04-01:
I haven't had a chance to dig into the verilog to determine whether or not this is already implemented this way, so if it is, I apologize. Executing an MSU-1 data seek to the same offset that is currently loaded should essentially be a no-op that immediately returns ready, or else if the read buffer isn't full, it should report busy but continue filling the buffer from its current position and not flush the buffer (i.e. don't actually perform a seek, but just report the buffer busy state as if you did). The reason for this is to allow re-syncing video DMA in the case of a buffer underflow by keeping track of the expected offset in the video player code and executing a seek every single frame. This method would not be feasible if seeking to the current offset actually performed a full seek with buffer flush because you'd incur the full seek and buffer time penalty every time, but if the cache was smart enough not to do that, the best case scenario of performing that seek when no desync had occurred would be essentially free, and so it would be able to continue normally.

mrehkopf, 2018-04-01:
Such optimizations are in place :) See src/msu1.c, function prepare_data (ll. 66)
Cache is only refilled if the address requested is outside the range represented by the current cache page. The busy flag is set by the FPGA and reset by the MCU upon completion so it will remain set for a very short time (probably a couple of S-CPU cycles only).

qwertymodo, 2018-04-01:
Fantastic. I'll try modifying my Chrono Trigger FMV player to implement this.  It's always a bummer trying to show it off only to have it crash halfway through into a seizure-inducing acid trip.

mrehkopf, 2018-04-02:
I don't recall the FMV patch well - are you using your own player or my example player? The latter got an update to hopefully help with erratic DMA behaviour.
If you are using your own - are you using HDMA to set forced blanking ($2100)? The DMA engine doesn't seem to like that very much, dropping DMA transfers at random...

closed 2018-04-02 by mrehkopf

qwertymodo, 2018-04-02:
I'm using my own player code which just sets the forced blanking on a busy loop that just reads the current scanline to wait for the start and end of the force blank period.  Probably not the most efficient way of doing it, but the CPU isn't doing anything else...

I've never had any issues on my smaller (~4GB) SD cards, so I just have one of those dedicated to my FMV games, but I'd like to figure out a solution for the larger cards, and I think this just might do it.