mrehkopf/sd2snes #98: Fire Emblem Issues

mbreeding3, 2018-06-29:
To whom it may concern,

I am trying attempting to play Fire Emblem: Genealogy of the Holy War on my SD2SNES. It works fine up until the prologues is about to begin, at which time the screen goes to black. Is there anything that I can do to fix this?

DieKatzchen, 2018-06-29:
Have you tried disabling in-game hooks? That's always the best place to start.

Optiroc, 2018-06-29:
And remember: if you're playing an old unfinished fan translation *all bets are off*. More often than not these are not properly tested on real hardware.

mbreeding3, 2018-06-29:
@DieKatzchen Yes, that is what I tried first. It still goes out right as the text introducing the Prologue appears on the screen. The text disappears, and then the map never appears. The music continues to play, but nothing happens. The rom works fine on snes9x, just not on SD2SNES.

@Optiroc Right; this is an original Japanese rom that has been patched with the project Naga translation, which is the only completed English translation of the game. Just to make sure that it was not something wrong with the patch, I tried the Japanese rom without the patch, and it still messes up at the same place.

Would it help you guys if I recorded a video of the issue?

mbreeding3, 2018-06-29:
Here is a clip of me running into the issue during a stream yesterday: https://clips.twitch.tv/BoredMiniatureOrcaNotATK

Ramsis-SNES, 2018-06-30:
Hi, I tried the original Japanese ROM just now, and it seems to work just fine. My guess would be that you might have a corrupt ROM file. Have you tried it on higan/bsnes? Also, you might want to post the MD5 sum of your file for comparison (don't forget to remove any 512-byte-long copier header, if present). Mine is d044f97be5fc00bd5e02c935f7e88058.

mbreeding3, 2018-06-30:
Here is my MD5 sum: d044f97be5fc00bd5e02c935f7e88058. It looks identical. Did you try going to the specific point in the game that I am talking about? My rom works perfectly up until then. And there are no issues at all when I play it on snes9x.

Ramsis-SNES, 2018-06-30:
Hi, yes, I started a new game and it got past the prologue screen without issue. Another thing to check would be your SD file system. If you have a different card on hand, try that as well. Also, remove/rename the SRM file of the game and see if that changes anything. Lastly, make sure you have all files from the current v1.8.0 firmware correctly installed in the sd2snes system folder. If if still doesn't work, I'd check out the SD2SNES diagnostics firmware on ikari's website as your PSRAM could be faulty. (Is FE4 the only game that causes trouble?)

mbreeding3, 2018-06-30:
Gotcha, that gives me some stuff to try. I did update to the v1.8.0 firmware last night, but I have not tried any of the other things that you mentioned. Yes, FE4 is the only game that causes trouble. 

Thank you for your help; I will try to delve further into this issue tonight. I have plenty of different things to try now, and for that reason I am going to go ahead and close this thread.

closed 2018-06-30 by mbreeding3