mrehkopf/sd2snes #100: Corrupt Srm file creation (from 0.1.7e)

josete2k, 2018-07-04:
If a game needs Sram file to work this file is created but fullfilled to FF and several glitches and crashes happen.

Deleting this corrupt file and putting a srm file from an emulator (snes9x) or launching the rom with 0.1.7c solve this issue...

0.1.8 has same bug.

mrehkopf, 2018-07-13:
This was actually a fix because many games didn't like uninitalized SRAM (or the contents from the previous game). Do you think it would be better to fill with some pseudo-random values? Might be worth testing.

josete2k, 2018-07-14:
An srm fullfilled to 00 seems to work, FF fails.

devinacker, 2018-07-14:
It was changed from 00 to FF to begin with (see #84) because 00 actually breaks several other games.

Which specific games are you having issues with? The only game I'm aware of having issues with using FF is Power Slide FX, but this change was made before Super FX games were supported anyway.

josete2k, 2018-07-14:
Chrono Trigger hangs at the pendulum intro (not always, but if not it freezes at the press start screen), Donkey Kong Country has glitches and Mario Kart Freezer randomly.

Ramsis-SNES, 2018-07-14:
Nintendo and Squaresoft (and I bet Rare, too) usually have checksums in their SRAM data to prevent the games from running should SRAM contents get corrupted. Out of curiosity, and because random crashes are hardly ever SRAM-related, I just tested all three games (DKC, Chrono Trigger, Super Mario Kart) with no SRM file in the /sd2snes/saves folder present, and they all work just fine.

mrehkopf, 2018-07-19:
The games mentioned work fine for me too (with FF initialized SRAM) so the underlying problem is probably something else...

josete2k, 2018-07-20:
But the issue is fixed with a 00 Sram file. 
The supercic key is set to pal on my cart, and I have a 1chip pal snes.

josete2k, 2018-08-03:
So, if FF it's not the problem...

Access speed maybe? I have issues with Tales of Phantasia even changing the srm file to 00.

0.1.7c is fine.

closed 2018-08-09 by josete2k