mrehkopf/sd2snes #99: Super CIC pair mode not fully working on 0.1.8

josete2k, 2018-07-04:
There are some issues in pair mode with this firmware.

0.1.7e is OK

TheLordkahless214, 2018-07-04:
Also having this issue, system Information reports "SuperCIC Detected, not used" "enhanced" or Pair mode last working correctly for me on 1.7 Preview 4 firmware.

mrehkopf, 2018-07-14:
Thanks for checking @Ramsis-SNES, works fine for me, too.
@josete2k @TheLordkahless214 Are you using a menu.bin from a previous firmware release (due to the custom theme hack)? Did you enable SuperCIC in the configuration menu -> SuperCIC settings?

Ramsis-SNES, 2018-07-14:
Das macht alles keinen Sinn.

TheLordkahless214, 2018-07-14:
I have tried a fresh installation, with new SD card and nothing but the new firmware folder, enabled SuperCIC from the configuration menu still shows "SuperCIC detected, not used"

mrehkopf, 2018-07-14:
Even after another power cycle?

TheLordkahless214, 2018-07-14:
Yes, and it still doesn't work I'm not sure what else to try

mrehkopf, 2018-07-14:
Might be a good idea to downgrade the firmware step by step and see if it actually still works with an older version (unless you did exactly that recently to determine it worked with 0.1.7b).

TheLordkahless214, 2018-07-14:
I updated my post above with photos, I did exactly that and it stops working at 1.7c

mrehkopf, 2018-07-14:
Thanks, I'll see what I can make of that.

TheLordkahless214, 2018-07-14:
You're welcome, and thanks for looking into this, hopefully it's an easy fix

mrehkopf, 2018-07-14:
What exact model is your SNES?

mrehkopf, 2018-07-14:
Found the problem.
In v0.1.7c the CPU clock was raised back from 86MHz to 96MHz. There is a count loop to determine the CIC status (it measures the toggle frequency of a status output pin on the SuperCIC key). It is a simple loop with a fixed number of iterations without any real time control.
The value read back depends on the CPU frequency of the sd2snes and also on the CIC clock frequency coming from the SNES.
The SNES CIC frequency can be:

- 4.000MHz (first models)
- 3.58MHz (NTSC models that run the CIC on the chroma subcarrier frequency)
- 3.54MHz (PAL models that run the CIC on a "fake" chroma subcarrier frequency)
- 3.072MHz (models that run the CIC on a clock output of the SPC700 sound chip)

The firmware needs at least 2500 toggles per measurement to detect SuperCIC enhanced mode.
On a "3.54MHz" sd2snes that I have, the measured toggles at 96MHz CPU frequency are around 2775.
I do not have a SuperCIC modded "3.072MHz" SNES but in proportion I am expecting the readout to be 2775 * 3.072 / 3.54 = **2408** which is less than 2500 and will prevent SuperCIC pair mode detection.
I need to make some more measurements and will then adjust the threshold value so it can detect reliably.

mrehkopf, 2018-07-16:
Fixed in b702b16

closed 2018-12-18 by mrehkopf