mrehkopf/sd2snes #101: "blue flash" when moving my character ...

theDarkFrenchy, 2018-07-17:
Hi !
First my configuration 

- 32 inch Samsung 1080p TV set,
- a super nt analogue (last firmware 4.5)
- an SD2SNES (last firmware 1.8.0).
- an SD with roms + MSU 1 versions

trouble : **"blue flash" when you move your character. you can see this flash when go up and down in the game. No blue flash when you stop character.**  the colour of "decor" is correct, only when you are not moving. It's "subtle but the " blue effect" is there !

Which games ? 
- secret of evemore (when you move your character in the jungle, up and down...)
- Shadowrun  (when you move your character, up and down, in the near the park, in the beginning of the game...)

I tried the normal rom  and also the MSU 1 version of the games. I didn't try other games (I own physically secret of evemore  and shadowrun..).
I post this issues on the super NT analogue  forums.
Since Sd2snes changed its firmware , I guess it's maybe coming from the new firmware ???

I don't know if I can "degrade" my sd2snes from 1.8.0 to previous one (0.1.7e ) ?

I didn't try to use the EXPERIMENTAL brightness patching for S-CPUN based consoles (1CHIP/Jr). 
My system is a super nt analogue  and not a 1CHIP (I sold mine .. so i can't compare or try on a real SNES...).

mrehkopf, 2018-07-17:
It's very likely to be an artifact of your TV's LCD screen (overshoot from overdrive to improve response time). sd2snes couldn't cause this by accident, much less in multiple games.

FiZiX, 2018-07-17:
> I tried the normal rom and also the MSU 1 version of the games. I didn't try other games (I own physically secret of evemore and shadowrun..).

Do you have the same issue when playing the physical copies?

theDarkFrenchy, 2018-07-17:
Ok, first I want to say : my Issue is gone. 
my Problem is SOLVED !
  
Yes I had the same issue with my physical copies !
Everything came from **the settings of  TV Set** : one of  friend who is "geek" , tweaked the settings , "in order to improve the picture quality" . 
I'm no geek regarding TV LCD, so.... 
My problems came from the the "**100 Hz Motion +**" setting... 
As soon as I reverted the setting to "OFF" everything was ok ! no more "blue flash"....  :-)

Sorry to bother you...   :-(  
Thanks for your help !

mrehkopf, 2018-07-17:
Case closed, issue closed ;-)

closed 2018-07-17 by mrehkopf