mrehkopf/sd2snes #102: S-RTC not detected by homebrew software

Ramsis-SNES, 2018-07-17:
In my [current homebrew RPG build](https://github.com/Ramsis-SNES/furryrpg/blob/ded9a32a167a2f4c0ffd32b871bb80a0877799a1/furryrpg.sfc), S-RTC is ignored, although it all works fine on higan (with a proper manifest.bml, of course). What am I doing wrong?

mrehkopf, 2018-07-17:
Cartridge type 02 does not enable S-RTC - try using 55 :)

Ramsis-SNES, 2018-07-18:
Ouch, totally forgot about the internal SNES ROM "header". ;-) Still no luck though ... What's even more strange, even higan ignores the RTC with cart type $55 (and without the manifest.bml). This is weird ...

Ramsis-SNES, 2018-07-19:
Converting the ROM to ExHiROM (48 Mbit) at least seems to satisfy icarus' heuristics, i. e. the ROM runs correctly on higan even without a manifest file. Gonna test the expanded ROM on my sd2snes later tonight and report back if it makes any change there, too. :)

Ramsis-SNES, 2018-07-19:
And sadly, sd2snes does not acknowledge the S-RTC with the 48 Mbit ROM (and cart type $55), either. :( So apparently, either one of the implementations of detecting S-RTC presence is bugged:

- higan doesn't detect S-RTC in cart type $55, unless ROM makeup is $35 (ExHiROM) (no manifest.bml in both cases)
- sd2snes never detects/supports S-RTC, regardless of cart type/ROM makeup bytes

According to Fullsnes,

> The sequence for setting, and then reading the time is

... something. Why on earth would I want to set the time **before** reading it, anyway?

I did a tracelog on the DKmono 2 intro, and a) it revealed a considerably different codeflow than what Fullsnes suggests, and b) in sd2snes's time.a65 (``gettime`` function), a #$0D command is issued to $2801 right before reading the time, which I duplicated for my own project (even though Fullsnes never mentions it) and which was what it made it work on higan in the first place.

This is silly (just like DKmono 2's intro tune). -_-

For testing purposes, I created a save with DKmono 2 on higan, and sure enough, it works just fine on sd2snes too (i. e., after the "Press Start" screen I get a save selection screen with the RTC running as it should).

Please don't get me wrong @mrehkopf -- not trying to blame your implementation here, but for a developer, it's just so frustrating to see completely opposite results on accurate emus like higan, and real hardware implementations like yours.

Apart from that, I still think that the problem sits in front of the workstation here. XD It'd be nice though if any of you SNES hardware gurus could chime in (BTW, is byuu even on GitHub?), and point me in the right direction.

Thanks, as always! :D

Ramsis-SNES, 2018-07-19:
Okay, quick and interesting correction/update:

- With a ROM bloated to 48 Mbit with Lunar Address, the RTC seems to work, but MSU1 is broken (no bird singing in area $0003)
- With an expanded ROM with my manual workaround (which might be faulty, as Fullsnes is extremely court on ExHiROM, but which is necessary as WLA DX doesn't support ExHiROM natively [yet?]), MSU1 still works but S-RTC is ignored.

Yay.

This is still very silly. :D

mrehkopf, 2018-07-19:
Well for one thing ExHiROM has two headers (one per A23 selected ROM) so make sure to set the cart type byte in both. (sd2snes prefers the one at 0x40ffc0 in the ROM file if it's present)

mrehkopf, 2018-07-19:
How can I check for S-RTC in the build you linked?

Ramsis-SNES, 2018-07-20:
> Well for one thing ExHiROM has two headers

Yep, I was/am aware of that.

> How can I check for S-RTC in the build you linked?

Sorry for my dumbness, but can you please rephrase the question?

BTW, I did some more testing, and it appears the Lunar Expand version 64M indeed behaves as described in my previous comment, but ... the Ramsis™ one actually behaves like this:

1. Upon loading the ROM, no RTC support, but full MSU1 support
2. Upon issuing a reset (either via LRSelStart or the Reset button when in-game hooks are disabled) both RTC and MSU1 support
3. Upon issuing a second reset ... Repeat from step 1.

This isn't just silly, this is absolutely stunning. :D

Here are the two different ROMs in case you don't believe me and/or still have the patience to try it out for yourself:

[https://manuloewe.de/snestuff/temp/furry64.7z](url)

Ramsis-SNES, 2018-07-22:
I did some (read: a ton) more testing/research, and by now I'm pretty sure it's all to do with the way how sd2snes detects MSU1/(and/or)SpecialChip™ presence. Proof-of-concept: Have two identcal ExHiROM ROMs with identical SRM files, but one has a 0-byte .MSU file and at least one PCM track to go with it while the other doesn't, and S-RTC support will oscillate between resets as described above.

Ramsis-SNES, 2018-07-23:
Okay, just to see if my presumption was right, I created a 0-byte MSU file for DKmono2, and tadaa, the RTC on the save selection screen remains at 0:0:0. :) After a reset, it seems to work (it took me to the stupid clock setting screen instead of the save selection though), and after another reset, it didn't work again (0:0:0).

mrehkopf, 2018-07-23:
Nice find! I assume that the reset handler for the MSU1 game loop doesn't reset the S-RTC state as opposed to the "normal" game loop...

Ramsis-SNES, 2018-07-23:
Should be an easy fix, right?

Ramsis-SNES, 2018-07-24:
Yay, thank you so much! :D

Ramsis-SNES, 2018-08-02:
Redguys SA-1 beta firmware incorporates your fix, and it's still kind of broken: When you launch the game from the menu, the RTC isn't detected/working. After 1+ resets, it works (and stays working).

Ramsis-SNES, 2018-12-23:
With the v1.9.0 release, it's still broken, as described in my last post. (Short version: On a 48 Mbit ExHiROM game with MSU-1 and S-RTC, the latter is only detected/used after at least one console reset either via in-game-hook-based button combo or plain old Reset button.)

mrehkopf, 2019-01-15:
I'll scrap the intermediate stuff and do the final FPGA side resetting... could you give me your current test ROM (that presumably somehow displays whether RTC was detected)?

Ramsis-SNES, 2019-01-16:
Sure, just grab the ROM from here: https://github.com/Ramsis-SNES/furryrpg/raw/master/furryrpg.sfc

On first start-up without an SRM file, the game will perform a CRC check, so you might have to wait for that to complete. As soon as the debug menu appears, it should immediately show the current time in the lower right corner of the screen.

mrehkopf, 2021-08-02:
Sorry I was behind the curve for a while. The time shows correctly for me (last tested on v1.11.0b1); are there still any issues?

Ramsis-SNES, 2021-08-02:
Thanks, it seems to have been working as expected since v1.10-ish, so I guess it's time to close this at last. ;-)

closed 2021-08-02 by Ramsis-SNES