mrehkopf/sd2snes #104: Tales of Phantasia radom freezes

josete2k, 2018-08-03:
I can play this game with no issues in 0.1.7c but in 1.8.0 it freezes at the intro or in battle.

Jap, Dejap and spanish translation have been tested with same results.

1chip pal supercic console and REV E2 -> REV H moded sd2snes.

Ramsis-SNES, 2018-08-03:
You do realize that with your ridiculous opening random issues just because YOU are having problems, you're effectively thwarting any real progress, do you?

josete2k, 2018-08-04:
I'm not the one with this issues.

0.1.7c is working fine.

Why newer firmwares doesn't?

mrehkopf, 2018-08-04:
So it breaks from v0.1.7c to v0.1.7d, correct? 
Did you try the diagnostic firmware already? https://sd2snes.de/blog/cool-stuff/sd2snes-diagnostics

mrehkopf, 2018-08-04:
I just noticed: Rev. E2? Is there a hand written serial number (s/n) on the circuit board? Or does it say manufactured by sd2snes.net on the circuit board?

josete2k, 2018-08-04:
The diagnostics are all passed except CIC not use, USB and DAC.

I had RTC issues once I moded the cart to REVH with Borti's pcb but they're gone by resoldering some components. I have the inductor and capacitor for REVF too.

As I said, the 0.1.7c is fully working and maybe not all pcbs are affected by this issue but I have some friends with same issues and same results. 

A fullfilled 00 sram may help but it's strange that some carts are affected a other aren't.

If it was a faulty Drive, it shouldn't work in older firmwares, I think... 

And about downgrading, there are some firmwares that can ve reverted for from an actual firmware. If I install 0.1.7 from 0.1.8sa1 I get (not always) an error code on screen, but if I install 0.1.6 and then 0.1.7 there are no problems.

mrehkopf, 2018-08-04:
What error message are you getting? Usually anything to anything should work fine...

Ramsis-SNES, 2018-08-04:
Maybe it's the new voltage regulator in the SNES ...?

josete2k, 2018-08-04:
Not sure... The 0.1.7c still works fine

josete2k, 2018-08-04:
The code (only happen with 0.1.7)

[img]https://www.subeimagenes.com/img/img-20180804-232737-249-1920822.jpg[/img]

I've also made a few videos with firmware 0.1.7e, 0.1.7c and the sa-1 beta.

Let me a few minutes to upload them

josete2k, 2018-08-04:
0.1.7e without any srm files in saves folder...

https://youtu.be/BVsK1N7xrWg

ToP starts the intro sequence, you ca hear some of the first Worlds and freezes. 

Star Ocean freezes at the Enix logo (its brighter just at the freeze).

josete2k, 2018-08-04:
0.1.7c

https://youtu.be/nvqzacrXNc0

All working fine. Even with the saves folder empty.

josete2k, 2018-08-04:
0.1.8 sa-1 beta

https://youtu.be/FXdmfQ35Z0o

srm files (00) in saves folder.

Star Ocean seems to be ok. 
Tales of Phantasia freezes at the intro (not shown in the video, sorry... it freezes at the end).

josete2k, 2018-08-05:
Yes I hope so.

Some people in forums had same problem with DKC and they fixed it with the 00 srm file.

Maybe is a faulty hardware, but as the 0.1.7c is working I think it can be fixed with software.

josete2k, 2018-08-05:
Little update.

I've tested two NTSC-U snes. 

First one has glitches on Mario rpg and Tales Refuse to run... 

As this con el nsole has glitches on sa-1 games I think that this is not the best testing console I can use.

The Second one is a snes mini tha has same problem that my 1chip pal.

https://youtu.be/NJTJlb2eMCk

Ramsis-SNES, 2018-08-05:
Oh really. :D

josete2k, 2018-08-05:
test_log

https://pastebin.com/26WjaYg1

ramapcsx2, 2018-08-06:
So it's the SD2SNES itself, probably related to what goes on on SRAM accesses.

Funky how that Tales hang causes that "dynamic audio driver" to not update. It creates a similar sound to really old SNES emulators that didn't implement this properly: Sound effects become a barking dog xD

josete2k, 2018-08-06:
Yes, it seems that it's an sd2snes issue... But where? 

I hope it could be fixed in new firmwares as the 0.1.7c is working fine.

PD: I've been playing SMRPG about 2 hours without any issue. So I asume my sd2snes is not "much" faulty. XD

Oh, I forgot to mention... Star Ocean freezes at all too, after the intro sequence.

mrehkopf, 2018-08-06:
[fpga_base.zip](https://github.com/mrehkopf/sd2snes/files/2264002/fpga_base.zip)
Does this work any better? (Copy over a v1.8.0 firmware install, overwrite fpga_base.bit in the sd2snes folder. Only for non-custom-chip games at the moment.)

josete2k, 2018-08-06:
@DoctorDan1986  thanks for the tip.
I've formatted the SD whith SDformatter and I have same results (in a few SD and microSD cards).

@mrehkopf, sorry same issue. I've placed a new sd2snes folder with 1.8.0 files and your fpga_base.bit... 

https://www.youtube.com/watch?v=W7jVl2mXZdA

The new srm file is all FF except offset 1F00: 9407070707070707070700303F00FFFF
I don't know if it helps.

josete2k, 2018-08-06:
Is there any way to disable the autosave sram like the MSU games do?

You know, Save to srm when reset is pressed. 

I think it should be a good thing to test.

josete2k, 2018-08-07:
Little update.

My friend magno has been checking the Sram access in Top and Star Ocean (he did both spanish translations for each game) and there are some sound variables stored in Sram that may cause the error (access speed maybe?). 

The strange thing is that the 0.1.7c hasn't got this issue.

I remember that ikari said that there were changes in the autosave feature in newer firmwares (0.1.7d?) so, is it possible that changes are affecting my sd2snes?

Is possible to add an "autosave option: ON/OFF" in the main menu? 

Thanks again.

mrehkopf, 2018-08-07:
@josete2k you could "disable" autosave by removing the SD Card after loading 🤓 but it would still keep monitoring the SRAM contents. But you could place a dummy file (e.g. "tales of phantasia.msu") in the same directory as your ROM (and with the real filename of the ROM). It can have any contents or even 0 bytes.
Then it will go to the MSU loop and stop monitoring the SRAM.

There was a change between v0.1.7c and v0.1.7e regarding the processing of the SNES ROM select line (you didn't tell me yet if it works with v0.1.7d ;) ) I reverted that change in the bitfile attached above.

I'm going to upload another bitfile later. There have been no changes in autosave or access speed, but the glitch filtering and SNES signal sampling points have been slightly adjusted. If that helps I'll blame your SNES for having extremely noisy address lines (or bad contacts). 🤣

josete2k, 2018-08-07:
Nice shot, I'll try tonight.

THANK YOU. 

XD remember that I have an usa mini tha has same problem...

mrehkopf, 2018-08-07:
Another thought: It could even be a contact problem on the sd2snes itself, maybe a suboptimal solder joint on some line somewhere between the slot contacts and FPGA, where the contact is "almost good". A high resistance could work well with the v0.1.7c timing but not with the later versions. If I had to guess I would suspect A23 or ROMSEL. ;)

mrehkopf, 2018-08-07:
Hypothesis: It doesn't get one of the address lines right for address decoding, and creates a transition between two addresses on the PSRAM bus mid-cycle, switching between a SaveRAM address (incorrect) and a ROM address (correct). The data delivered by PSRAM then also changes mid-cycle. Based on the data contents of the incorrect memory location this could work better or worse during the transition, which would explain why the SRAM contents matter at all in this equation.

josete2k, 2018-08-07:
Checked lines and they seem all fine, contacts are clean and shiny.

But...msu file has solved the issue!!!!!

Tested with the last 1.0.8 SA-1 V3

josete2k, 2018-08-08:
Tested 0.1.7d

Games not launching :( even with sram or not. NONE OF THEM!!!!

Note that pair cic mode is not working... it seems that system crashes just when pressing A button in the rom.

https://www.youtube.com/watch?v=POpLz4byI48

Back to 0.1.7c (or 1.8.0 sa1v4 with msu file) and they're working again.

I think it's time to think about to get a new cart in the everdrive store. Even so, I think it's a "problem" that should be solved.

Thanks for your support, ikari.

mrehkopf, 2018-08-08:
Interesting. v0.1.7d has a slower ROM access cycle (8 clocks delay instead of 7 in both v0.1.7c and e)
Anyway here's a new bitfile to try. This combines the faster ROM access cycle with the pre-0.1.7e edge detection settings.
[fpga_base2.zip](https://github.com/mrehkopf/sd2snes/files/2272330/fpga_base2.zip)

josete2k, 2018-08-08:
YEAHHHH it works!!!!!!

josete2k, 2018-08-13:
After a few days of testing I can confirm that this fpga_base solves the issue (even in SA-1 beta firmwares).

THANK YOU!!!

closed 2018-12-18 by mrehkopf