mrehkopf/sd2snes #107: some games are not working when is this fixed

wolfenkraft, 2018-08-22:
title says it all, expecting be resolved in one week (08/31/2018)

borti4938, 2018-08-22:
title says nothing. This issue is spam and rediculous imo. reported

wolfenkraft, 2018-08-22:
there are games that are not working, see the official website and stop hating

magu, 2018-08-22:
🤦‍♂️

Ereza, 2018-08-22:
Why don't you fix the issues yourself? I expect you to have fixed all of them by 08/31/2018...

Really, you should close this issue immediately and go learn some manners.

wolfenkraft, 2018-08-22:
you've lost a customer, I'm getting a everdrive now
and the community of it is way friendlier, thanks

magu, 2018-08-22:
If that solves your issue: good for you. To each his own.

As helpful advice, you need some practice in filing actionable support issues though.

wolfenkraft, 2018-08-22:
and you need to do your work and enable support for the unsupported games instead of posting hateful comments

borti4938, 2018-08-22:
You should read first and get to know what the sd2snes project is first instead of opening rediculous issues. Then you may get usefull answers.
sd2snes is an open source project and free to everyone in the community.

> you've lost a customer, I'm getting a everdrive now
> and the community of it is way friendlier, thanks
Ikari is the creator and a hobbiest, not a seller. I guess Krikzz has lost a customer and won back the same customer in one step. And btw: the list of unsupported games for the Super Everdrive is a bit larger than the one for the sd2snes. Also the hardware is not as powerful.

wolfenkraft, 2018-08-22:
please tell me how to reach ikari, I want to talk to himself about this issue and how you handle issues here

sylverb, 2018-08-22:
OMG, is this real ? O_o

magu, 2018-08-22:
*looks around for hidden camera*
🙈🙉🙊

hedgehog1029, 2018-08-22:
this is the funniest issue I've seen in my entire life
how did this person become so entitled???
what do they think they're gonna say to the project leader???

DoctorDan1986, 2018-08-22:
*deep breath*

Wolfenkrant, you are not being particularly clear.  What games do you claim are not working?  Have you updated the firmware to the most recent version?  Are you aware that some games are KNOWN to not work yet?

As we cannot see inside your head to divine your meaning, you will have to be explicit when it comes to the issues you are having.  Take a few minutes, think about how best to describe the issue, and then write it out.  The community is more than willing to lend a hand, but you have to meet us at least halfway.

devinacker, 2018-08-22:
> Wolfenkrant, you are not being particularly clear. What games do you claim are not working? Have you updated the firmware to the most recent version? Are you aware that some games are KNOWN to not work yet?

Based on the discussion so far, I'm pretty sure they're aware of this, and yet are demanding that the single maintainer of the project add support for _every SNES game ever released_ by the end of next week.

Or else they'll, you know, take their business elsewhere and buy an Everdrive (from the same manufacturer) and be able to play even fewer games instead.

Ramsis-SNES, 2018-08-22:
Ri-di-cu-lous. Wolfenkrant, get your Wolfen ass outta here and see a doctor. Please!

magu, 2018-08-22:
@doctordan I see your point, but I don't think he even has any actual problems, since he doesn't appear to even have bought an sd2snes yet, he just seems grumpy that not all games work.

Don't get me wrong here, but I just expect your current endeavour to be fruitless.

magu, 2018-08-22:
Help yourself ...

[image]
[image]

wolfenkraft, 2018-08-22:
i've tried to update the firmware now nothing is working. this is outrageous

justindarc, 2018-08-22:
Your abrasive, entitled, smug attitude is outrageous.

joeydumont, 2018-08-22:
@wolfenkraft "Nothing is working" is not very descriptive. If the sd2snes doesn't boot at all, there is a good chance you didn't put the firmware files in the right place on the SD card. If the sd2snes boots but specific games won't start, please state the problematic games and we'll go from there.

wolfenkraft, 2018-08-22:
I tried to update the firmware now it says FLASH CHECK ERROR (clone?)

magu, 2018-08-22:
Where did you buy your sd2snes from?

wolfenkraft, 2018-08-22:
I don't know why that's important to solve my issue. I got it at a Russian website where I buy all my video games. For a pretty good price krikkz can't compete with

magu, 2018-08-22:
Because that points to it being a cheap clone, which might explain some problems you are having.

magu, 2018-08-22:
Krikkz charges the price he does mainly not out of greed, but because quality costs money to produce.

Also, the error you get "FLASH CHECK ERROR (clone?)" does seem like the firmware detects something funky with your (probably) clone hardware.

wolfenkraft, 2018-08-22:
deelete this

closed 2018-08-22 by wolfenkraft