mrehkopf/sd2snes #111: DKC 3 random black screens with v1.8.0, works on 0.1.7d

robinhedwards, 2018-09-12:
I'm getting random hangs with a black screen (the music continues playing) with Donkey Kong Country 3 (USA) on the current firmware (v1.8.0). Typically the hang occurs before I even manage to play the first level.

Other games seem to work fine on the SD2SNES (though I've only tested a handful), and it also passes all the tests in the diagnostic firmware.

Console is a Super Famicom (2/1/3).

I discovered that by reverting the firmware to 0.1.7d, the game plays fine - I played as far as the second level with no problems. 0.1.7e fails like 1.8.0, so the problem seems to have been introduced between 0.1.7d & e, and persists to the current firmware.

Interesting the same ROM plays fine with v1.8.0 on a Super NT with the same SD2SNES.

I'm happy enough to leave my cartridge on 0.1.7d, but thought it was worthwhile reporting this issue, which may be related to #104.

mrehkopf, 2018-09-12:
Does it help if you use the fpga_base.bit from the file fpga_base2.zip that I attached to the issue that you linked?

robinhedwards, 2018-09-12:
Yes - that seems to fix the issue with v1.8.0 - Only had a quick playthough up to the 2nd level, but I don't think I've ever got that far with the stock v1.8.0 firmware.

closed 2018-12-18 by mrehkopf