mrehkopf/sd2snes #113: Reset to Menu not working at the moment (04. Oct. 2018)

borti4938, 2018-10-05:
I made a build from current sources yesterday. It seems that the reset to menu does not work at the moment - neither short reset (with option short reset to menu enabled). nor long reset nor double reset. Using the reset to menu using in-game hooks gives a black screen (game stops, sound is still running). In this state an additional tap on the reset button returns my sd2snes back to the menu.

I wasn't able to quickly figure out what causes the behaviour. I hope that this is reproducible and this report helps to figure out what's wrong!
Or did I do something wrong?

James-F2, 2018-10-06:
Try to compile redguys sa1b fork, it should be exactly the same up to this point:
https://github.com/RedGuyyyy/sd2snes/tree/sa1b

The latest sa1b fork works correctly with more than a few consoles that I and a few others have tried.

borti4938, 2018-10-06:
Tested today on another console. This time it's working fine.
- console where it does not work anymore: Ultra16 modded (serial 87)
- console where it works: SuperCIC modded
Both consoles SNSP-1Chip-02