mrehkopf/sd2snes #114: Rev. H PCB data?

DevLaTron, 2018-10-23:
Hi,
 I was looking for a Revision H. PCB KiCad or Gerber set, but the repository only contains up to Rev. F. I'd like to construct my own SD2SNES, and of I'd like the OpAmp circuit on board if possible.

Could someone point me in the right direction for the Files?

devinacker, 2018-10-23:
No idea about the origin, but this fork has Altium (but not KiCad) files for Rev. H:
https://github.com/gargan26/sd2snes/tree/develop/pcb/altium/rev_h

DevLaTron, 2018-10-25:
@devinacker Thanks, I managed to pull the board and create Gerber X2 files from it. It seems to be correct from what I can read.

Sadly I haven't managed to convert it to a sensible KiCAD version to create a pull request from. :-/

DevLaTron, 2018-12-27:
Hey together... it's been a while...

since my last post, I've spent quite some time with the schematics of the SD2SNES and its PCB layout.

I've "kobbled" together a Rev. K. branch (WIP, feel free to have a look and tell me how bad it is): 

https://github.com/DevLaTron/sd2snes/tree/devlatron/RevKWork

It contains:
- Added OpAmp TS922 + Resistors to AUDIO_R and AUDIO_L output.
- GNDed pin 28 of MC to set HardwareID.
- Changed PCB layout to fit Hirose DM3AT MicroSD  instead of large SD Cards. Rerouted traces to fit new size.
- Fixed and added Library and Footprint files, also converted files to KiCAD 5 format where appropriate.
- Fixed broken Footprints all over the board because of Version and library hickups.
- Make it pass KiCAD DRC.

Open:
- Postion component descriptions in a more sensible way.
- Final logic and design checks. (E.g: Is this sane?)

closed 2023-03-07 by mrehkopf