mrehkopf/sd2snes #117: Random delay (3-9 sec.) when the Reset button is pressed in SA-1 games

Ramsis-SNES, 2018-12-23:
Title says it all (firmware used: official v1.9.0). This happens both on unmodified and SuperCIC-enhanced consoles, regardless to the in-game hook setting.

mrehkopf, 2018-12-25:
It's probably the reset hook (for phase correction) waiting for the firmware to respond, which in the first seconds of game operation is now still busy with ROM CRC calculation to decide on partial SRAM monitoring.
Does this happen to a freshly booted game only or also when it's been running for a while?

mrehkopf, 2021-08-02:
Closed due to lack of feedback, feel free to re-open or just reply with any new input. :)

closed 2021-08-02 by mrehkopf

Ramsis-SNES, 2021-08-02:
Oops, sorry. No idea why I never got back to your question. :-( I can't test it right now but from memory, I don't think I observed any reset delay in v1.10.x, so no worries. ;-)