mrehkopf/sd2snes #118: Allow folder-based MSU-1 (instead of ROM-based)

april, 2019-01-07:
Currently to use MSU-1, you have the .pcm files with the same name, e.g.:

```
somerom.sfc
somerom-1.pcm
somerom-2.pcm
```

While this is fine most of the time, it's really painful for randomly generated ROMs, such as the Link to the Past randomizer. Those roms all have filenames like this:

```
ALttP - VT_no-glitches-30_normal-open_randomized-ganon_29MQDmZ0GD.sfc
ALttP - VT_no-glitches-30_normal-open_randomized-ganon_0kGZKmpEMm.sfc
ALttP - VT_no-glitches-30_normal-inverted_randomized-ganon_E2v7nnWJMq.sfc
```

And so on. To play those, I have to pop out the SD card, copy one over, rename it to static name, and pop the SD card back in. It would be great if I could have a dozen different ROMs in the same folder that all used the same PCM files, regardless of their name. It would allow me to copy over a dozen ROMs at once without having to mess with filenames. So you could have:

```
ALttP - VT_no-glitches-30_normal-open_randomized-ganon_29MQDmZ0GD.sfc
ALttP - VT_no-glitches-30_normal-open_randomized-ganon_0kGZKmpEMm.sfc
ALttP - VT_no-glitches-30_normal-inverted_randomized-ganon_E2v7nnWJMq.sfc
1.pcm
2.pcm
3.pcm
4.pcm
```

And no matter which SFC file I opened up, it would use the same PCM files. It would save me (and a ton of randomizer players who use SD2SNES's) a bunch of time.

Thanks so much!

codemann8, 2019-08-08:
+1 for this request.  I was about to post this same thing. @mrehkopf

Cinaeth, 2023-05-09:
Not a full solution, but this is what you're looking for: https://github.com/tcprescott/RomLoader - setup the yaml to recognize the file name of the randomized rom, eg lttp. Then you can setup to have it write to the specific folder and name convention for each MSU. As long as your SD card has a connection to the PC it's great ( such as mini USB from console or pack to USB on PC - you might have this setup for auto tracking or multi worlds)

codemann8, 2023-05-09:
It's not a solution at all! This isn't an issue with the delivery method of the ROM, it's an issue of MSU1 requiring the ROM filename prefix to match the prefix of the pcm files. This poses some potential problems. This makes it so that every seed you play it overwrites the previous seed's SRM file. (< Not great if you want to participate in Async Multiworlds, since you can't have multiple games active at once)