mrehkopf/sd2snes #123: Multiple Build Issues, missing files

DevLaTron, 2019-05-20:
With the latest update of the file structure, a couple of issues arise:

- Syntax Error in main.v (Closing braces on line 98? Runaway parentheses?)
- Compiled binaries aren't included in .gitignore for utils/
- mem2lorom doesn't compile on Ubuntu, needs include file (C99 vs POSIX)
- Paths are hardcoded into multiple Makefiles.
- main.ngd is skipped on smartxplorer in Makefile

- SA1 for Xilinx is missing source files referenced in project: srtc.v upd77c25v updtest_tf.v
- SDD1 for Xilinx is missing source files referenced in project: updtest_tf.v
- MINI for Xilinx is missing source files referenced in project: data.v

- CX4 MK3 doesn't build: my_dcm.qip not found? (leads to snes_dcm instantiates undefined entity "my_dcm")

- SNES Menu doesn't build: Missing cheat.a65 referenced in Makefile.

DevLaTron, 2019-05-20:
CX4 actually builds. For some reason there is a .\cx4ram.coe reference... that won't work on linux... changing it to ./cx4ram.coe will make the core generate correctly and thus build.

freelancer42, 2020-02-18:
Did you ever get this to work? I'm trying to set up a build environment for Mk3 development, which is proving... difficult (any tips for setting up a working Linux environment for this?). Biggest issue right now though is that the damn menu (which is my main concern) won't compile because of that missing cheat.a65 file. Any idea where I could possible get a hold of it?

closed 2021-08-02 by mrehkopf