mrehkopf/sd2snes #126: Super Mario World - Graphics Glitch During Iggy & Larry Battles

shiggypoo, 2019-06-10:
**_Super Mario World_** as well as **_Super Mario All Stars + SMW_** both exhibit this glitch during Iggy’s battle.  It also happens during Larry's (same scene with added fireballs)

It seems that at least two other users on Krikzz forum experienced this.  It happens intermittently, one user claims disabling in-game hooks fixed it but it happens off and on for me either way... more often occurring than not.

**Original forum post:** http://krikzz.com/forum/index.php?topic=7421.15

**Video of issue:** https://youtu.be/AuiWbzxyiqM

- Using latest 1.10.3 firmware on a brand new SD2SNES Pro

- In-game hooks are off (tried with them on, one it didn't happen, but subsequent events it occurred)

- SNES is 2/1/3

- SD2SNES Pro passes diagnostic firmware and the console passes the burn-in rev D test.

- Recently cleaned pin connector of console with a credit card, 99% isopropyl on a microfibre cloth and canned air.

- ROMs sourced from a no-intro compilation.

mrehkopf, 2019-06-25:
Hmm. Could it be a 1CHIP console despite the serial number?

shiggypoo, 2019-07-02:
Sorry for the delay, I didn’t get a notification and just happened to check in and see your response over the weekend.  I just took it apart and confirmed it is not a 1CHIP model. Board model # is SNS-CPU-GPM-01.

mrehkopf, 2019-07-11:
Now I've seen it too on my GPM-02, doesn't seem to happen every time. Would you check if turning on the setting "Reset patch for clock phase" in the In-game settings helps?

shiggypoo, 2019-07-11:
Aha! Enabling it seems to have done the trick. For me I would experience the glitch around 80% of the time. Now it’s been three times and has not occurred.

I notice that this setting is kept off by default.  Will keeping it enabled have any adverse side effects with other games?

I’ll report back if it happens again. Thanks for your help! :-)

closed 2020-04-16 by shiggypoo