mrehkopf/sd2snes #127: Hotkey issue SD2SNES Pro

MarcoMartins86, 2019-06-26:
I've bought recently an SD2SNES Pro and installed the latest firmware 1.10.3, so I have never tried with older ones, I am using SuperNt with the latest firmware also.
The behavior I think is an issue is as follow:

1. Was playing an MSU-1 game, The Legend of Zelda: A Link to the Past (not sure if this is relevant)
1. I was aware of the saving issues so I was trying to reset the game by using L+R+Select+Start as described [here](https://sd2snes.de/blog/compatibility/in-game-hook-compatibility) but it wasn't working, also tried the others shortcuts without any luck.
1. So I tried the soft reset button on the console which saved the game, and it suffices for me. Nevertheless, when I come back to play the game I was going up and ended pressing Select+Up which did a soft reset of the game and I lost my progress.
1. Next, I have discovered that if the Select+Up is pressed in less than 2s +/- it did a soft reset, otherwise it would call the cartridge menu.
1. Next, I have changed to all the possible combinations of setting in (Main Menu (X button) -> Configuration -> In-game Settings -> In-game hook) and none seems to make a difference.
The advertised keys combination never work and Select+Up is always working. The only setting that would work was the one where if I reset the game it should go to the cartridge menu instead.

Can someone else confirm this behavior?

mrehkopf, 2019-06-27:
Select+Up is a key combination coming directly from the SuperNt which resets the console. so that would explain why that always works.
Regarding the sd2snes in-game buttons not working, that would be subject to further investigation. Are you using any sort of unusual controller?

MarcoMartins86, 2019-06-27:
I have both for so little time I didn't know that, but yes just confirmed that the Select+Up is from SuperNt. As for the SD2SNES key combinations not working, I've tested with the 8bitdo SN30 and SN30 Pro, both with the latest firmware.

MarcoMartins86, 2019-06-27:
Ok did some more tests and concluded that they are working but I need to have all the keys pressed for like 4s to trigger them and I thought it should be immediate. So this can be closed.

closed 2019-06-27 by MarcoMartins86