mrehkopf/sd2snes #128: Support for sfroms

piratk, 2019-06-28:
Some emulators and the SNES Mini use "sfrom" rom file format. It would be nice if sd2nes would support these out of the box, so there would be no need to use tools to convert sfrom to SFC.

mrehkopf, 2020-10-13:
This might be of limited use, sfrom files seem to have their BRR audio samples converted to PCM and stored separately so it won't be able to run with audio on a real SNES. But I'm not sure what percentage of sfroms actually uses this format?

piratk, 2020-10-14:
To be honest, I've only seen these roms in the use case I described, when dumping them from one of the mini-consoles. But that said, as there are two mini out there, [with 26 roms in total](https://en.wikipedia.org/wiki/Super_NES_Classic_Edition#Games).

mrehkopf, 2020-10-14:
https://www.reddit.com/r/miniSNES/comments/7p2lzw/discussion_how_do_i_convert_sfrom_to_sfc/
Here DarkAkuma points out that the SFC Mini dumps must have audio converted back from PCM to BRR (the format that the SNES audio chip understands) which entails a loss of sound quality. Also the conversion itself does involve some operations that are computationally expensive on the sd2snes's embedded hardware which will slow down loading a lot.
All things considered I don't think the outcome would be satisfactory, nor justify the effort...