mrehkopf/sd2snes #132: SuperCIC 1.13 Not working

LittleRainGames, 2019-11-28:
I've been trying to get it working for a couple days now.

My custom cart wasn't working, so I got an original cic from SMW to see if it was something else, but my cart started working after that.
I also tested the supercic in my SMW cart to no avail.

I just compiled v1.10, and my cic chip is now working. 
I didn't test any versions between 1.13 and 1.10,
Also I'm using an NTSC console if that makes any difference, cant get console version at the moment, but can if you would like me to.
Better take it off the SD2SNES website!

mrehkopf, 2019-12-03:
Hmm, that's interesting because there are no changes to supercic-key.asm between v1.10.2 and v1.10.3. I had a similar problem when prototyping SuperCIC on the MK3 PCBs but it was due to pull-ups I forgot to disable on the I/O pins of the ARM MCU connected to the CIC pins.

LittleRainGames, 2019-12-13:
Hmm, well maybe its worth noting the v1.10.3 that wasn't working for me was downloaded from the SD2Snes website, while the other ones I compiled.

mrehkopf, 2021-08-02:
I updated the precompiled hex archive at https://sd2snes.de/files/supercic.zip which is confirmed working so far. Feel free to re-open if you encounter any issues :)

closed 2021-08-02 by mrehkopf