mrehkopf/sd2snes #135: please help pic 12f629 ta very expensive. build 12f675?

darleiv, 2020-03-01:
how do I recompile super cic to use the pic 12f675 ??? don't you have this compilation? the 12f629 is too expensive

magu, 2020-03-01:
Using the lock.asm as template and modifying the key file should get you there.
[supercic_12f675.patch.txt](https://github.com/mrehkopf/sd2snes/files/4272229/supercic_12f675.patch.txt)

magu, 2020-03-01:
Haven't got the slightest clue if it works, but it compiles fine. YMMV.

darleiv, 2020-03-01:
> Haven't got the slightest clue if it works, but it compiles fine. YMMV.

I only know how to use vhdl and C # c ++ [] [] BUT I don't know anything
does make file create the file in hex?

darleiv, 2020-03-01:
> Haven't got the slightest clue if it works, but it compiles fine. YMMV.

if you can send the file I test it here at px675

magu, 2020-03-01:
[supercic-key.hex.zip](https://github.com/mrehkopf/sd2snes/files/4272273/supercic-key.hex.zip)

darleiv, 2020-03-01:
at first no work kkk

darleiv, 2020-03-01:
but thanks for the help

darleiv, 2020-03-01:
PIC12F629 and PIC12F675 are 8-Pin Flash-Based 8-Bit CMOS Microcontrollers. The PIC12F629 and PIC12F675 devices are identical, except the PIC12F675 has a 10-bit A/D converter. They come in 8-pin PDIP, SOIC, MLF-S and DFN packages.

mrehkopf, 2020-03-02:
I think for 12F675 the analog pins are enabled as a default and must be set as digital GPIO pins in some place where register bank 1 is selected (below a `bsf STATUS, RP0` instruction):
```
clrf ANSEL
```

darleiv, 2020-03-02:
mrehkopf digital conversion to digital analog. because the 12f675 is not standard for using the firmware already compiled. unfortunately i already spoke with some programmers only that i would be able to do it does not have the tools installed. if you feel like trying to get people to have this option. I am willing to test the compilation. I have all pics of several microchip models here. I tried to compile in MPLAB IDE unfortunately the compilation is not accepted hehe - many people asked me for this on my super IC site 12f675

magu, 2020-03-02:
Just get https://gputils.sourceforge.io and you should be able to compile it yourself.

darleiv, 2020-03-04:
> I think for 12F675 the analog pins are enabled as a default and must be set as digital GPIO pins in some place where register bank 1 is selected (somewhere below a `bsf STATUS, RP0` instruction):
> 
> ```
> clrf ANSEL
> ```
too complicated to do? if you can recompile it makes it available to people

cetics, 2020-05-08:
here my hex converted.
https://atariage.com/forums/topic/306540-supercic-for-12f675-here/

mrehkopf, 2021-08-02:
Commit 02873c0594171b95c75514379068b26cd784e0c1 adds support for more PIC models to the compilation.
Pre-compiled hex files can be found here: https://sd2snes.de/files/supercic.zip

closed 2021-08-02 by mrehkopf

darleiv, 2021-11-15:
mrehkopf pic 683 and 675 nowork in GPM one chip 03