mrehkopf/sd2snes #137: Bug while DMA from msu to vram.

gizaha, 2020-05-23:
Test1: I have vram graphics in msu file, address $34d3. I dma them to vram but sends only $b2d length (34d3+b2d=4000). 
Test2: I move the data to msu address $4000 and dma is working ok.

Other times (ingame), i set address to $2000-$2003 wait for data busy, dma data and (sometimes) it reads from wrong (advance) address. 

Both problems at vblank. I tried f-blank and close hdma, no difference.
No MSU music is playing to lighten the load.
US sd2snes 1.10.3, pal snes 2/1/3
SD card has avg 1.275ms  and max 1.696ms

mrehkopf, 2020-06-01:
Hi,
1. what happens after copying $b2d bytes?
2. can you estimate how much the address is advanced after the busy loop?
Could you provide example code?

gizaha, 2020-06-02:
1) Dma stops.
2) Because one picture is a thousand words, check it yourself to see the
buffer overflow.
Use left right and keys.
Jump from address 200 to 4000 (keys X-A) to see what happens. If you press
A again, its correct.
If you have access to uart for read printf you will gain time.

;sd2snes test rom

hirom
;vectors
	org $ffc0
		db "sd2snes test rom"
	org $ffd5
		db $31		;3.58mhz hirom map mode 21
		db 0		;without coprocessor
		db $9		;32mbit (9=4mbit, c=32)
		db 0		;no ram
		db $14		;region=other
		db $33		;fixed value

	org $ffea
		dw nmi_code

	org $fffc
		dw reset

org $8000
reset:
;set cpu native mode
	sei
	lda #$01
	sta $4200
	stz $420c
	lda #$8f
	sta $2100
	clc
	xce
;setup stack
	rep #$10
	ldx #$01ff
	txs
;setup bank
	phk
	plb
;setup direct page
	rep #$20
	lda #$0000
	tcd

;setup screen mode
	sep #$30
	lda #$01
	sta $2105		;screen mode
	stz $2106		;mosaic
	lda #$f8/2
	sta $2107
	lda #$f0/2
	sta $2108
	sta $2109
	
	stz $210b		;bg1/2 char address
	stz $210c		;bg3/4 char address	
	stz $210d	
	stz $210d		;bg1 hor scroll
	stz $210e
	stz $210e		;bg1 vertical scroll
	
	lda #$80
	sta $2115		;video port control (2118/9 inc)
	
	stz $2121		;cgram address
	stz $2122		;first color in palette=black
	stz $2122
	
	stz $2123		;bg12 window disable
	stz $2124		;bg34 window disable
	stz $2125		;oam window disable
	
	stz $212a		;mask
	stz $212b		;mask
	lda #$01
	sta $212c		;bg1 mainscreen
	sta $212d		;subscreen
	stz $212e		;mainscreen window disable
	stz $212f		;subscreen window disable
	
	stz $2130		;color addition
	lda #$01
	sta $2131		;color math
	lda #$e0
	sta $2132		;bg color
	stz $2133		;disable pseudohires
;set linear tilemap
	rep #$20
	lda #$f800/2
	sta $2116
	lda #$2000
-	sta $2118
	inc
	cmp #$2400
	bne -
;enable nmi
	ldx #$80
	stx $4200
;reset ram positions	
	stz $10
	stz $12
	stz $14
	stz $16
	stz $18
;loop forever
- 	;nop
	;bra -

--	lda $18
	beq --
	
	
	;wait line 0
-	ldx $4212
	bmi -
	;read keypresses
	rep #$20
	stz $14
	lda $4218
	sta $10			;current keypress
	cmp $12			;previous keypress
	beq +
	sta $12
	sta $14			;one frame keypress
+
	stz $18
	bra --
	
	
	
	

nmi_code:
	sep #$20
	lda $4210		;clear nmi flag
	lda #$8f
	sta $2100		;close beam
	lda #$81
	sta $4200		;80=enable nmi, 1=joypad read

	inc $18
	

.check_keypresses	;that latched in previous frame nmi
	rep #$20
	lda $14
	bne +
	jmp .exit
	
	;RIGHT
+	cmp #$100		
	bne +
	lda $16
	cmp.w #(.keys_end-.keys)-2/2
	beq ++
	inc $16
	bra ++
	
+	;LEFT
	cmp #$200	
	bne +
	lda $16
	beq ++
	dec $16
	bra ++
	
	;L
+	cmp #$0020
	bne +
	tdc
	bra +++	
	;R
+	cmp #$0010
	bne +
	lda #$4200
	bra +++
	;Y
+	cmp #$4000
	bne +
	lda #$100
	bra +++	
	;X
+	cmp #$0040
	bne +
	lda #$200
	bra +++	
	;A
+	cmp #$0080
	bne +
	lda #$4000
	bra +++
	;B
+	cmp #$8000
	bne .exit
	lda #$3400
	bra +++
	
;set msu address	
++	lda $16
	asl
	tax
	lda .keys,x
+++	sta $2000
	stz $2002
;dma tiles
	stz $2116
	lda #$1809		;read from msu file
	sta $4300
	lda #$2001
	sta $4302
	ldx #$00
	stx $4304
	lda #$3000
	sta $4305
-	ldx $2000
	bmi -
	ldx #$01
	stx $420b
	

	
.exit
	ldx #$0f
	stx $2100		;open beam
	rti
	
.keys
	dw 0, $800, $1000, $1800, $2000, $2800, $3000, $3800, $4000, $4800
.keys_end

gizaha, 2020-06-02:
I forgot:
1) When the bug occurs, dma stops. If i read 4305 is zero. Also it's NOT at
"data busy".
So i couldn't detect when the bug occurs to make a workaround. I cannot
predict how advance is, i see advance frames from next videos but couldn't
calculate the address.

gizaha, 2020-06-23:
It's not DMA problem but sd2snes msu buffer. 
Instead of DMA, i upload byte by byte reading 2001 and writing to 2118-2119. I have the same result.
This took 3 frames, so it's not read speed error. msu1.c has bug at creating it's data buffer.

mrehkopf, 2020-10-10:
Indeed there was a bug where the read offset status flag wasn't updated correctly on a buffer hit optimization case.
47bd526f should fix this.

gizaha, 2021-04-25:
Any update on this? 
How can i use the 47bd526 changes? Shall i compile the firmware myself?
I work at sfa2 msu1 and this bug is holding me back again.

mrehkopf, 2021-08-02:
Fixed in v1.11.0b1.
On a semi-related note, some performance improvements may still be required for data seeking in general.
Unfortunately MSU1 doesn't provide a way to tell the hardware how much data the program needs in advance so it always buffers the entire 16384 bytes which makes it uneconomical to try and load from multiple offsets per VBlank.
I have some ideas for optimization but want to sort out bugs of the current beta release first.

closed 2021-08-02 by mrehkopf