mrehkopf/sd2snes #139: Super Mario RPG has graphical issues in the world environment on SD2SNES PRo on 2/1/3 SNES (possibly 1chip)

Goldanas, 2020-07-01:
[image]

Random graphical artifacts like those in the picture can appear in the world environment. They don't always appear, and they don't always appear in the same place (so they do seemingly have somewhat random and errant behavior). An easy way to spawn them in is to press X and go to the menu and then exit the menu, change place, and then repeat to see if they show. They seem almost random, but they might be based on your screen position and when it loads the background and how.

If you're familiar with the game, you'll notice them appearing about 50 percent of the time, sometimes online 1 or 2, and sometimes as many as 10.

I have a real original SMRPG cart, and these artifacts do not appear when playing on that cart.

leoleotte, 2020-08-11:
I can vouch for this one too, and inside some rooms you can get stuck because the collision data is different from the visual one, and you can't reach a room's exit for example...

Happens a lot when opening the menu

mrehkopf, 2020-08-11:
I can acknowledge this, been there before. There's been some discussion of it in 71e7ec5 where I posted an alternative bit file for the Mk.3 / Pro: https://github.com/mrehkopf/sd2snes/commit/71e7ec54098e38d4c7ea38617b2810675ac07fa3#commitcomment-36408651
Could you check that out and report back?

Goldanas, 2020-08-12:
That straight up fixed it. Thanks!

closed 2020-08-12 by Goldanas