mrehkopf/sd2snes #149: Mortal Kombat II hangs, black screen, important info here for the future.

Nashismo, 2020-12-21:
Dear mrehkopf:

Here is the deal, your Firmware 1.10.3 is the greatest, what happens to me is that when using any firmware version besides yours, which in this case would be Furious version with savestate support, then this game, Mortal Kombat II (REV1, USA), hangs and displays all sort of issues.

The problem is fixed if reverted to your firmware 1.10.3!

I could not find a Github to inform Furious about this, the problem started when he added save state support. I am telling you so you consider to test this "specific game" if you ever add this feature in your main firmware.

The game hangs even if a disable in game hooks, cheats and savestates on Furious firmware, in yours it works as it should.

My best regards! And thank you for your amazing work.
Nashismo.

mrehkopf, 2021-01-05:
Thank you!! I will take this into account when preparing the first beta of the new release.

Nashismo, 2021-01-06:
Thank you mrehkopf, in fact it is the only game that has ever crashed on me on the Furious firmware, but it does it consistently. I think it is a timing issue probably, but it would be good for testing.

I aslo tested this same ROM on a cheap chinese everdrive, and it worked perfectly for many hours.

Ok, won't take away from your time anymore, thanks again.

mrehkopf, 2021-07-08:
I did some testing on the current beta 1 build and it seems to work well. The build is finished, I'm writing the post for it as we speak. Please look forward to it :)

mrehkopf, 2021-08-02:
@Nashismo can you check if it works better on v1.11.0b1?

Nashismo, 2021-08-02:
> 
> 
> @Nashismo can you check if it works better on v1.11.0b1?

Yes, I did not see this message. I will be checking today! Thank you Mrehkopf!

Nashismo, 2021-08-02:
Yes!! Mortal Kombat 2 working great here!! I played for about an hour.

I also saw a memorial for Near. It seems it only plays one time. I do not want to get into scandal, but Near's death hasn't been confirmed yet or proved to have actually happened. Hopefully Near only did this to disapear as on online persona.

Thank you dear Mrehkopf!

closed 2021-08-04 by Nashismo