mrehkopf/sd2snes #150: The Auto Region patch doesn't work on EU console

Telecrab, 2021-04-13:
**SNES model:** SNSP-001A(FRG) - 1CHIP - PAL.
**Cartridge:** SD2SNES Mk.II Rev. H
**Firmware**: v1.10.3

**Steps to reproduce**

1. Turn on "Auto Region patch" in settings.
2. Try to launch Kirby’s Dream Land 3.

**Expected result**

Game starts.

**Actual result**

Wrong region message is displayed.

**Additional info**

Resetting/power cycling the console after the setting change doesn't work.

mrehkopf, 2021-08-02:
It should generally work. I've had one sd2snes where it didn't work but it was a fairly old build (Rev.E2). The cause was wrong values of the ferrite arrays on the data lines (RA103, RA104).
Can you check if it works better with the recent beta release (v1.11.0b1)? Otherwise would be cool if I could send you a bunch of FPGA files with tweaked timings to see if any of them helps.

ICEknigh7, 2021-08-03:
I can confirm that it doesn't work on the latest version, trying games such as "Ganbare Daiku no Gen-san".

mrehkopf, 2021-08-10:
I looked at it through the logic analyzer and indeed the patched data appears quite late on the bus, maybe too late for some CPUs. I'll prepare an FPGA file for testing that responds a bit earlier on the data bus.

Telecrab, 2021-09-12:
Big thank you for the update and sorry to hear about Near.

Checked this FW https://sd2snes.de/files/sd2snes_firmware_v1.11.0b1.zip
"Kirby’s Dream Land 3" works now. :heavy_check_mark:
"Super Metroid (Japan, USA) (En,Ja)" doesn't work. :x:
"Peace Keepers, The (USA)" doesn't work. :x: (Worked before, but hanged up at the copyright/logo screen half of the time. Didn't have any bugs during gameplay though.)

Overall compatibility is improved compared to the previous version. Many more games started to work. For example "Ghost Chaser Densei (Japan)".

wirehack7, 2023-05-26:
Sorry to open this again. But the error is back: even with auto patch I get the region error message from the cartridge (testing Secret of Mana (Germany) dump).
My SuperNT runs as NTSC, SOM is PAL: getting wrong region error

Used firmware version: 1.11.0

ICEknigh7, 2023-05-26:
Also doesn't work when trying PAL games on a Super Famicom.

Romain80200, 2023-06-05:
Hello,
I copy my message from another thread:

"Hello, please look at this thread:
https://github.com/mrehkopf/sd2snes/issues/186

Use this file:
https://github.com/mrehkopf/sd2snes/files/10778271/issue186-try2.zip

It solves the problem with cic and the missing leds."

ICEknigh7, 2023-06-05:
> Use this file:
https://github.com/mrehkopf/sd2snes/files/10778271/issue186-try2.zip

Just tried `Super Mario All-Stars + Super Mario World (Europe)` with this version and it still displays the region lockout message.

Now it also randomly fails to reboot when I press Reset, with all leds blinking at once.

Romain80200, 2023-06-05:
Maybe try the one at thebottom of this thread:

https://github.com/mrehkopf/sd2snes/issues/201

bitrot-alpha, 2023-08-30:
**Console:** 2/1/3 (USA SNES)
**Cartridge:** Chinese "Rev. X"

Tried 1.10.3 FW. Terranigma (EUR) worked after toggling Reset Patch, SuperCIC, and of course Auto Region Patching.
Then tried 1.11.0 FW. Tried same toggling of settings. Terranigma then booted up fine. I'm guessing timing is still not *perfect*, especially with dubious quality :smiley: of Chinese SD2SNES cartridges..

Always a good idea to try the basic troubleshooting steps:
- Power off and wait a little while before powering on again
- Reseat cartridge
- Clean cartridge contacts
- Blow out SNES cartridge slot with some kind of compressed air

bitrot-alpha, 2023-08-31:
Posting MD5 checksums of the files I have since some games have multiple revisions.
@ICEknigh7 can you tell us what version of SFC you have? The numbers I posted in my last comment are from the "System Information" menu on the SD2SNES.

All the games I tested work fine. I tried disabling Auto Region patch just to check the files I have aren't pre-patched, and DKC3 correctly showed the wrong region message on startup.

Donkey Kong Country 3 (EUR) :heavy_check_mark: 
MD5: `fc3333af2f93d7dd0a4190a01d49a69e`  Donkey Kong Country 3 - Dixie Kong's Double Trouble! (Europe) (En,Fr,De).sfc

Secret of Mana (EUR) :heavy_check_mark: 
MD5: `0194f74c93b0edf3e4a7672a448572ad`  Secret of Mana (Europe).sfc

Super Mario All Stars combo ROM (EUR) :heavy_check_mark:
MD5: `4941387e96ed2841ee3c2de6d1899509`  Super Mario All-Stars and Super Mario World (Europe).sfc
Problem with the background flashing pretty badly on SMB 1985, but it boots without the out of region message. Probably due to the fact that my SNES and TV are USA 60hz (No SuperCIC on SNES side)

ICEknigh7, 2023-08-31:
My current console is also 2/1/3 and DKC3 always shows the region lock screen. The difference I can see is that it's a Japanese one, maybe this only works on US systems?