mrehkopf/sd2snes #156: [v1.11.0b1] Small blemish in "Favorite games" menu

Ramsis-SNES, 2021-08-06:
If you press B to go back from here (image taken from https://github.com/mrehkopf/sd2snes/pull/134) ...

[image]

... the menu will flicker slightly, and the selection bar will be placed on the topmost entry in the favorite games list instead on the one the context menu was opened for. (Normally, all sub-menus close without any flickering, and the selection bar is placed on the previously selected element.)

Not a big deal at all, I know, but it just caught my eye. ;-)

Ramsis-SNES, 2023-01-29:
Awww, I was hoping for a fix for this in the final version. :p Oh well. ;-)

freelancer42, 2023-01-29:
This one's probably my fault. I implemented the favorites feature after digging through the SD2SNES code without any prior knowledge of how it worked, so I might've picked the wrong subroutine for something. Sorry :)
I'm glad people find it useful though!

mrehkopf, 2023-01-29:
Heh, forgot about it. I was really too focused on getting the critical stuff to work on Rev.D but I think it was necessary...
Will fix it for 1.11.1 ;)

mrehkopf, 2023-02-19:
Please give it a try :) [menu.zip](https://github.com/mrehkopf/sd2snes/files/10778196/menu.zip)

Ramsis-SNES, 2023-02-20:
> Please give it a try :) [menu.zip](https://github.com/mrehkopf/sd2snes/files/10778196/menu.zip)

Works just fine now, thanks a lot, ikari! :D

PR #199 (2023-02-19): Fix refresh/cursor issues in "Favorite games" menu
Fixes #156

UI stuff is always the most painful...
- There was no way to return outside of the submenu system whether a
  function was executed so a `listsel_dirty` flag was introduced to force
  refresh
- Same for the `listsel_backup` variable to keep track of the selected list
  item before entering the submenu
- Redrawing the list after removing a favorite resulted in some flickering so
  an additional `screen_dma_disable` flag was introduced to inhibit
  screen refresh (WRAM->VRAM DMA) until the redraw was finished.
  This feature can be u [... cut]
commit 48fecdb5 (2023-02-19): Fix refresh/cursor issues in "Favorite games" menu
Fixes #156

UI stuff is always the most painful...
- There was no way to return outside of the submenu system whether a
  function was executed so a `listsel_dirty` flag was introduced to force
  refresh
- Same for the `listsel_backup` variable
- Redrawing the list after removing a favorite resulted in some flickering so
  an additional `screen_dma_disable` flag was introduced to inhibit
  screen refresh (WRAM->VRAM DMA) until the redraw was finished.
  This feature can be useful for other redraw operations.
closed 2023-02-21 by mrehkopf