mrehkopf/sd2snes #157: [v1.11.0b1 regression] Game lock-up during tunnel scene in Star Fox/Starwing on SuperCIC-modded PAL SNES

RAKtheUndead, 2021-08-07:
Version: v1.11.0 beta 1
SD2SNES Model: SD2SNES Rev. J

Using the most recently released beta version of the SD2SNES firmware, whenever I attempt to start a new game in both Star Fox (NTSC-U) or Starwing (PAL), the graphics freeze at various points during the tunnel scene prior to starting the game, with the sound continuing, but following this, the game locks up entirely. This happens regardless of which level I choose. 

I can enter the Training Mode and play that with no apparent problems. I also have not seen the same issues with v1.10.3 and have subsequently tested this version again to ensure that these issues are not occurring.

mrehkopf, 2021-08-08:
Does that also happen with in-game hook disabled?

RAKtheUndead, 2021-08-08:
It doesn't appear to and I hadn't used in-game hook with v1.10.3 before, so I think in this case that I can safely assume that this is an issue with IGH.

Is there a list of games which have known issues with IGH aside from this one? https://sd2snes.de/blog/compatibility/in-game-hook-compatibility

pkerga, 2021-11-21:
Same issue for me with Starfox on my SFC (graphics freeze at various points during the tunnel scene).
But it works well with in-game hook disabled

closed 2023-02-05 by mrehkopf