mrehkopf/sd2snes #158: Bad video output since Firmware 1.10.3! Only now I realize!! (Interesting phenomenon)

Nashismo, 2021-08-08:
Dear mrehkopf:

I will try to make it short. I have a Super Famicom for 1 year now. Since I have it, I have only used the SD2SNES with Firmware 1.10.3.

Here is the thing, ever since I've got it I have had issues with video quality, because I have this interference noise in the image output. I have gone crazy, replaced all the capacitors, the voltage regulator and I even added the 470uf capacitor in the new voltage regulator!

I indeed eliminated some jailbars, the sound even seems a bit better, but that "noise" in the image, I could never get rid of!!

I have the 1st model of the Super Famicom. Reading all over the Internet I found people with the same issue on the same model of the Super Famicom.

So, the last thing I did was removing the RF Modulator from the SNES. It did not fix the problem, I cleaned the AV Multi out, it did not fix the problem.

BUT! While I was at it, I thought HEY! I should play an original USA game on it now that the console is disassembled (because you cannot play USA games on a Japanese SNES). And LOW AND BEHOLD, the games played perfectly!! No noise in the image! No strange interference in the video!!

It was the SD2SNES!!

I put the same games on the SD2SNES, and I was BACK with the same interference and strange wavy lines in the video output!

So I said, what if I go BACK in the firmware?? (I had never used any other than 1.10.3 mind you).

I installed Firmware 1.10.2 and BAM!! All my problems where FIXED!!

NO MORE WAVY LINES!! I almost cried :)

Dear mrehkopf, don't know why no one else has come out to you with this issue, but it's true!!

If you ask me to test anything please let me know!! One thing I noticed in the newest firmwares is that the "games" image looks "brighter" than in an original game or firmware 1.10.2.

Thank you for reading mrehkopf and I hope something can be done for future releases of the firmware, I would also be happy to assist you in anything,

Best regards,
Nashismo

ICEknigh7, 2021-08-10:
Hey, so I'm not the only one with picture problems on a Super Famicom?

I have a 1/1/1 Super Famicom with four rubber feet (first batch?) which also starts getting wavy output after a few minutes of using the SD2SNES, and after some more time even the colors go wrong if using composite video (they don't change through RGB).

Here's some pics I took of the issue: https://imgur.com/a/xXcRzBn

Observations:
-I'm using an official Mega Drive power adapter.
-Same problems happen even with a 2A power adapter (a 3A one is on the way).
-This doesn't happen with regular official cartridges (played Yoshi's Island for a while and looked perfect), except for a small amount of interference after using the Super Game Boy for some time.
-I've opened (and cleaned) the console and nothing inside looked busted.
-My PAL 2/1/3 SNES doesn't show any of these problems, no matter how long I've been playing with the SD2SNES.

If it's an effect of the oldest Super Famicom models struggling with the SD2SNES' power consumption, maybe some kind of an "energy saving" mode could help fixing this, even if it's only for games which don't use special chips?

mrehkopf, 2021-08-10:
Well that doesn't look so great... what revision of sd2snes/FXPAK are you both using?

Nashismo, 2021-08-10:
> 
> 
> Well that doesn't look so great... what revision of sd2snes/FXPAK are you both using?

I opened my SD2SNES and it says Revision X, made in 2020. And like I said, now that I am using your Firmware 1.10.2, everything seems to be working great!

If you want to I can test any other new revisions of your Firmware, so that anyone using my SD2SNES version/Super Famicom Model 1 combo, doesn't have problems in the future :)

Nashismo, 2021-08-10:
> 
> 
> Hey, so I'm not the only one with picture problems on a Super Famicom?
> 
> I have a 1/1/1 Super Famicom with four rubber feet (first batch?) which also starts getting wavy output after a few minutes of using the SD2SNES, and after some more time even the colors go wrong if using composite video (they don't change through RGB).
> 
> Here's some pics I took of the issue: https://imgur.com/a/xXcRzBn
> 
> Observations:
> -I'm using an official Mega Drive power adapter.
> -Same problems happen even with a 2A power adapter (a 3A one is on the way).
> -This doesn't happen with regular official cartridges (played Yoshi's Island for a while and looked perfect), except for a small amount of interference after using the Super Game Boy for some time.
> -I've opened (and cleaned) the console and nothing inside looked busted.
> -My PAL 2/1/3 SNES doesn't show any of these problems, no matter how long I've been playing with the SD2SNES.
> 
> If it's an effect of the oldest Super Famicom models struggling with the SD2SNES' power consumption, maybe some kind of an "energy saving" mode could help fixing this, even if it's only for games which don't use special chips?
> 
> EDIT: Sadly, downgrading to 1.10.2 didn't get rid of the problem in my case: https://imgur.com/a/Th8w5ub

That's weird, it is not what I was having! Yours is too much and it seems like a different problem. In fact I remember I once bought a crappy chinese Power supply for my Gamecube, and those wavy lines look EXACTLY like my Gamecube did using that PSU.

Check the PSU! It might be original, but it might be going bad!

FiZiX, 2021-08-10:
> Hey, so I'm not the only one with picture problems on a Super Famicom?
>
> I have a 1/1/1 Super Famicom with four rubber feet (first batch?) which also starts getting wavy output after a few minutes of [...]

Sounds like your SNES needs a re-cap; especially because it starts after a few minutes of use. My guess is the SD2SNES is putting additional strain on capacitors that are already beginning to fail due to its higher power consumption (compared to original games).

ICEknigh7, 2021-08-10:
> > Well that doesn't look so great... what revision of sd2snes/FXPAK are you both using?
> 
> I opened my SD2SNES and it says Revision X, made in 2020. 

Same as mine, got it second hand but it seems to be like new.

> Also you say "I've opened (and cleaned) the console and nothing inside looked busted" Wrong, nothing looked busted in mine either, but the sound was horrible and the jailbars were bad

Thing is, everything looks and sounds perfect even after hours playing an official cartridge of Yoshi's Island.

Nashismo, 2021-08-10:
> Same as mine, got it second hand but it seems to be like new.
>
>
> Thing is, everything looks and sounds perfect even after hours playing an official cartridge of Yoshi's Island.

I understand, but I assure you that replacing capacitors and voltage regulator will help tremendously. It is actually easy when you look at consoles like the genesis that has like 40 capacitors to replace! The SNES has like 8 ot 10 if I remember correctly. Your capacitors might "work" but that doesn't mean they are in their best years, specially when working with a device they were never intended to work with. At least replace the voltage regulator, I would start there, because it is a power issue you have.

symbios24, 2021-08-30:
Hi,as far as i know the sd2snes revision x is a chinese clone not the original,always buy from the official site.

ICEknigh7, 2021-08-31:
> Hi,as far as i know the sd2snes revision x is a chinese clone not the original,always buy from the official site.

You could at least show some evidence that the version sold elsewhere would work better in these cases. Else, your reply isn't helping anyone here at all.

symbios24, 2021-08-31:
Is this enough for you to believe its not authentic?? 

https://m.aliexpress.com/i/4001147366106.html

https://m.aliexpress.com/item/4000377747886.html

Nashismo, 2021-08-31:
> 
> 
> > Hi,as far as i know the sd2snes revision x is a chinese clone not the original,always buy from the official site.
> 
> You could at least show some evidence that the version sold elsewhere would work better in these cases. Else, your reply isn't helping anyone here at all.

Actually you are into something here, if we could only have someone with the 1st Super Famicom model with a NON revision X firmware. Then we could see if it is the SD2SNES revision or the SNES model.

Nonetheless, I do understand that this is a non-Krikzz version, but that does NOT mean mrehkopf cannot help us! That I know of, SD2SNES was open sourced at some point and that's why other people started to make their own versions.

If this was Krikzz forum I would understand. I have also purchased Krikzz stuff, directly from Ukraine, but this is the only cart that is not made by Krikzz, excuses or not, I live in south america, so I prefer to eat and if I can save 100 dollars I am going to.

symbios24, 2021-08-31:
No problem,but still its a low quality cart,unknown how it performs in special chip games,and has an old fpga i thing cyclone 2 you could just wait for black friday.

Nashismo, 2021-08-31:
> 
> 
> No problem,but still its a low quality cart,unknown how it performs in special chip games,and has an old fpga i thing cyclone 2 you could just wait for black friday.

Do you make things up on a regular basis? Yet, you do not know and just speak. The cart is well built, the soldering is great, has voltage regulators built in, it has a Cyclone 2 yes, but so what? The Spartan on the original SD2SNES is slower, actually the Cyclone 2 can be 2x times faster than the Spartan according to Intel:
https://www.intel.cn/content/dam/www/programmable/us/en/pdfs/literature/wp/wp_dsp_comp.pdf

Here you have pictures of the Revision X SD2SNES https://photos.google.com/share/AF1QipOScTDE7bzsQg9ZnM17pv0LuFj3-IpFpFM4Eg2MOQFLBKzMl8aDUZXcLkwb9dmV8g?key=dFhOVTJMb1JkMF8tS3JvTGhTZUpEU29qa2JHQTln

All special chips work perfectly fine, even the MSU-1 games, everything works exceptionally well. So, you speak in ignorance. But now you now.

ICEknigh7, 2021-08-31:
> if we could only have someone with the 1st Super Famicom model with a NON revision X firmware. Then we could see if it is the SD2SNES revision or the SNES model.

Exactly what I meant, that would be helpful.

I myself didn't know what version it was since I got it second hand, but it might be irrelevant to the case anyway.

symbios24, 2021-08-31:
Then we should all buy clones 😂

Nashismo, 2021-08-31:
> 
> 
> Im talking in general if you think you are insulted its not my problem, we should all buy clones coz nashismo says are better joy

Insulted? I don't understand your thinking pattern honestly. You are wrong and I am right, as simply as that, I don't feel insulted, I called you what you are, an ignorant, you seem to feel insulted for that it seems. But I don't care, because we are not here for that, let's try to stay on topic, this is still not a chat, we are here to ask mrehkopf for assistance.

Nashismo, 2021-08-31:
> Exactly what I meant, that would be helpful.
>
> I myself didn't know what version it was since I got it second hand, but it might be irrelevant to the case anyway.

It clearly has to do with this 1st models of SNES. Remember also you said your PAL version worked with no issues. We are at the mercy of mrehkopf for now, hopefully he can take our issue in consideration for his next firmware update.

symbios24, 2021-08-31:
Try the beta firmware maybe it works

https://sd2snes.de/blog/archives/1157

Nashismo, 2021-08-31:
> 
> 
> Try the beta firmware maybe it works
> 
> https://sd2snes.de/blog/archives/1157

That I remember I already tried that one, because  I got the Near message at the beginning. Again, another thing that my mind cannot grasp as Near's death hasn't been confirmed by any authority, only the Internet. In this irrational times I try to keep myself always investigating for real facts.

symbios24, 2021-08-31:
Did you try the tape fix?

https://krikzz.com/forum/index.php?topic=9725.0

Nashismo, 2021-08-31:
> 
> 
> Who is that Near?
> 
> Did you try the tape fix?
> 
> https://krikzz.com/forum/index.php?topic=9725.0

No I haven't tried the tape thing, thank you, I don't believe it has to do with my issue, but it is always good to try new things! Near is another name for Byuu, the creator of the MSU-1 patching and the developer of the Higan emulator (Highly accurate SNES emu).

He was supposedly a resident in Japan, he also seems to be a person with X gender (I don't understand those things, blame my nationality if you want) and he supposedly commited suicide while speaking on the phone with a friend playing a specific song he liked. 
I won't go into details, but nothing makes any sense in his supposed death. And there are no Japanese authority that confirms it.

My theory is the "Near" did this (made this up) to "disappear" from public/internet, as he had many problems with lots of people. I hope that is the case because he was extremely talented.

symbios24, 2021-08-31:
A senario from movie who knows what happen
Tried older firmwares?

Nashismo, 2021-08-31:
> 
> 
> A senario from movie who knows what happen
> Tried older firmwares?

Yes! I tried Frimware 1.10.2 and my SNES works without issue there! And that is one of the many reasons why I think my SD2SNES version is a good quality one, clone or not. Actually when I showed you the pics is for you to check how good the quality is on this specific revision X. I cannot speak for other "clone" revisions.

And yes, seems like the scenario from a movie, it took me quite some time to find that short info I gave you on "Near" supposed death, after investigating A LOT. It happens with me, sometimes I go out of my very rational mindset, and my "guts" tells me when something "might be" bullshit hahhaha, and that is why I investigated, because Byuu has nothing to do with me, but, it caught my atention and after investigating, I knew why I had the "gut" feeling about it.

symbios24, 2021-08-31:
better to stay with the older firmware for a while if it works ok,Maybe he owns a lot of money and decide to go in the shadows 😂
Do you have a retrogen adapter on snes?

Nashismo, 2021-08-31:
Yes that firmware 1.10.2 works with all its features on my SD2SNES so no need to panic for me, at least for now hahahhha. 

And no I don't have that adapter, it's to play Genesis games right? I just got a Genesis recently and modded it for "jailbar free" RGB. If you want tips for that, I'll be glad to give them to you (in other words, I don't need the Retrogen hehe). Best regards.

symbios24, 2021-08-31:
I dont have a megadrive thats why i took the retrogen its was cheaper for me,and saves space they my retro corner,the image for composite is quite good and the compatibility its high too from all the games i tried only altered beast was showing artifacts,chakan rocks!!! Works flawless with mega ed x3

symbios24, 2021-08-31:
No problem.continue your issue sorry for the off topic thing

ICEknigh7, 2021-11-14:
> No problem.continue your whatever issue I doubt you find any answers

Nice edit history, from polite to rude in less than two weeks.

ICEknigh7, 2021-12-06:
> At least replace the voltage regulator, I would start there, because it is a power issue you have.

Just did that and all the video interference problems are gone, thanks for the suggestion! :smile:

Nashismo, 2021-12-07:
> > At least replace the voltage regulator, I would start there, because it is a power issue you have.
> 
> Just did that and all the video interference problems are gone, thanks for the suggestion! smile

REALLY!! I am so glad for you ICEknigh7! But my question is, did you replace the voltage regulators on the console or the small voltage regulators on the SD2SNES itself? 

Thank you, and I am really glad you took notice of my advice!

ICEknigh7, 2021-12-07:
> did you replace the voltage regulators on the console or the small voltage regulators on the SD2SNES itself?

Just replaced the one inside the Super Famicom with a L78S05CV which cost me less than 0,2€, so it was also a pretty cheap fix.

Nashismo, 2021-12-08:
> > did you replace the voltage regulators on the console or the small voltage regulators on the SD2SNES itself?
> 
> Just replaced the one inside the Super Famicom with a L78S05CV which cost me less than 0,2€, so it was also a pretty cheap fix.

That's great! Then my last question would be, what Firmware are you using? 1.10.2 or later ones? Ok, take care then :)

ICEknigh7, 2021-12-08:
I'm using "1.10.3_sgb07".

Nashismo, 2021-12-08:
> I'm using "1.10.3_sgb07".

 Ok, very interesting. This info will help me :)

fjsj, 2023-03-25:
FYI, I have video noise on 3 different SNES, 2 of them 1-chip-01, 1 APU. I have a sd2snes rev. x (the Chinese one). Tried 3 different PSUs for SNES, the OEM one gives me less noise, but still there. I use OSSC, so I tried a different PSU for it too, noise got even worse. Tried different monitors, different wall plugs, nothing solved it. Original cartridges don't have that noise. The noise is present even on sd2snes menu.

The noise is mostly horizontal and diagonal, random, and it's not the checkering noise from composite or sync-on-composite video cables. I use a csync properly shielded SCART.

Noise is on firmware 1.10.2, 1.10.3, 1.11.0, perhaps a bit worse in 1.11.0.

Finally, I got a SNES Mini with RGB-bypass and full-recap: no noise in it! The only remaining noise I have is the [1-chip ghosting issue](https://www.retrorgb.com/1chipghosting.html), which is also present in original cartridges (I tested with SMW).

Therefore, as others stated in the thread, it's perhaps old capacitors, voltage regulators, and other components can cause noise only on sd2snes.

Nashismo, 2023-03-25:
> FYI, I have video noise on 3 different SNES, 2 of them 1-chip-01, 1 APU. I have a sd2snes rev. x (the Chinese one). Tried 3 different PSUs for SNES, the OEM one gives me less noise, but still there. I use OSSC, so I tried a different PSU for it too, noise got even worse. Tried different monitors, different wall plugs, nothing solved it. Original cartridges don't have that noise. The noise is present even on sd2snes menu.
> 
> The noise is mostly horizontal and diagonal, random, and it's not the checkering noise from composite or sync-on-composite video cables. I use a csync properly shielded SCART.
> 
> Noise is on firmware 1.10.2, 1.10.3, 1.11.0, perhaps a bit worse in 1.11.0.
> 
> Finally, I got a SNES Mini with RGB-bypass and full-recap: no noise in it! The only remaining noise I have is the [1-chip ghosting issue](https://www.retrorgb.com/1chipghosting.html), which is also present in original cartridges (I tested with SMW).
> 
> Therefore, as others stated in the thread, perhaps old capacitors, voltage regulators, and other components can cause noise only on sd2snes.

I am sorry but as you describe, which I thank you so much, all consoles have problems with this cart!! On the opposite your RGB mod was the one that got rid of the noise. It is not because any capacitor or voltage regulator in bad condition! You see I also fixed my issue by using and RGB to Component box that my brother had, somehow this box gets rid of the noise, as others didn't. BUT, the noise will always be an issue on a all unmodified SNES's using these chinese SD2SNES's. This thread is useful to advice people to try to stay away from these chinese products if they can.

lanotlab, 2023-06-22:
Guys, I was having the same problem. Tried all firmwares and no luck. Turns out it was my psu cable that was faulty.
Now everything works fine even for long playing hours.
It's a cheap china version btw