mrehkopf/sd2snes #159: [v1.11.0 beta] Final Fantasy 3 US equipment screen flickers

Jeffersonian, 2021-08-21:
FF3 US rom has a graphical issue when you bring up the equipment screen (screen flickers like crazy, menu can’t be viewed).  I downgraded to v1.10.3 and the issue doesn’t occur.  Tested on a Super Nt.
[image]

mrehkopf, 2021-08-21:
Unfortunately this is a regression caused by the introduction of the save state handler which is called from the in-game hook. When the in-game hook consumes too much SNES CPU time the HDMA setup of the FF3/6 menu does not finish in time. This is why it was broken in early firmwares before I optimized the in-game hook code.
For now it should work fine with in-game save states disabled but I acknowledge that the save state handler is in dire need for CPU time optimization. I will begin optimizing it from v1.11.1.
Right now I'm on summer vacation so expect fxpak activities to continue in about two weeks ;)

CrispyBuns, 2023-04-12:
if you disable in-game hooks it fixes this issue. Also in japanese version and the uncencored and MSU-1 versions

jfineberg, 2023-08-21:
Just wanted to ping here and see if this was still on your radar, @mrehkopf !

mrehkopf, 2023-08-21:
Yes, I will attempt a fix after the next release; for the time being the only solution is to turn off in-game savestates, sorry :(