mrehkopf/sd2snes #161: MSU1 games with motion video glitches on 1.11.0b1

kimbapslice, 2021-08-31:
MSU1 games with motion video glitches on 1.11.0b1 and not on 1.10.3. I am using 2/1/3 snes (ntsc/usa). Switch back to 1.10.3 works fine.

kimbapslice, 2021-09-05:
I tested Super Road Blaster today on 3 fully working (as far as I can tell) USA snes with configuration 2/1/3 snes (ntsc/usa). I have disabled all in-game menu and gamesaves. Snes burn-in test v1.03 reports all pass on all 3 systems.

Results for Super Road Blaster with 1.11.0b1 video glitches on all 3 snes.
Results for Super Road Blaster with 1.10.3 NO video glitches on all 3 snes.

My sd2snes is an old official E1b revision that I have upgraded to revision H.

manwithouties, 2022-04-15:
I know this post is old... well, I'm experiencing the same issue. Have you found a solution?
I don't want to keep using 1.10.3 because I love the save state and SGB features.

kimbapslice, 2022-04-16:
Ok, so here's an update on further testing. I recently acquired a snes 1chip-01 and snes mini/jr. Both the 1chip-01 and snes mini/jr report 2/1/3 configuration. Both are NTSC/USA revisions.

Super Road Blaster with 1.11.0b1 on snes 1chip-01 = video glitches
Super Road Blaster with 1.11.0b1 on **snes mini/jr = normal, correct video**

New findings!

manwithouties, 2022-04-17:
@kimbapslice Thanks for your reply and testing! So, no easy fix! I tried manipulating some files on the sd2snes folder with no success. Anyway, it's a firmware issue and needs a fix... let's hope developers can solve it on next update. Meanwhile I'm using two SD cards, one with 1.11.0 and other one with 1.10.3.

mrehkopf, 2023-02-27:
SNES mini/jr and 1chip are technically identical. Could either of you provide a recording of the glitches?
Do the glitches recover at all on the MSU1 video demos (e.g. Akira)?
Are the glitches still present in the 1.11.0 final? Can you let the SD card access time measurement complete on the System Information screen and post the results?

mrehkopf, 2023-03-07:
Thank you 🙇‍♂️ So far I've been unable to reproduce the issue here. The only thing that caused intermittent glitches was a very slow card with a max access time of 11ms... However there seems to be an issue with bus timing on 1.11.0 (#204, #184), maybe the underlying cause is similar.
Can you run Illusion of Gaia and check for flickering of the Nintendo/Enix logos and corrupted graphics?

kimbapslice, 2023-03-07:
Hello,

So today I tested the 1.11.0 release with 2 snes (1chip-01, and not a 1chip), 2 snes jr. All four systems are 2/1/3, running composite video. First tested with my sd2snes upgraded to 1.11.0

Results on 1.11.0:
Snes (not 1chip, stock), snes 1chip-01(brightness fix+vertical band fix) = MSU1 videos GLITCH
Snes JR (stock): MSU1 works
Snes JR (rgb mod, brightness fix): MSU1 works

Results on firmware 1.10.3
MSU1 videos work on all consoles

I hope this helps!

Thank you so much for your work!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

https://user-images.githubusercontent.com/58018661/223561377-7df4587b-527a-4802-884b-4cf5c32f426f.mp4

https://user-images.githubusercontent.com/58018661/223561402-c653811a-9a31-4b3b-8d7f-ea301367dafa.mp4

mrehkopf, 2023-03-07:
Thanks! Looks like it's missing one byte of data pointer increment at the beginning of each DMA transfer... will need to check how recent changes might have caused that.

mrehkopf, 2023-03-08:
Can you try this (only for OG sd2snes)? Just replace the fpga_base.bit in your sd2snes folder with the one from the zip file.
It's not a final fix, just changed the way MSU address is incremented to close in on the issue.
[fpga_base_msu1_address_inc.zip](https://github.com/mrehkopf/sd2snes/files/10925532/fpga_base_msu1_address_inc.zip)

kimbapslice, 2023-03-08:
Wonderful news! The fpga_base.bit did the trick and now firmware 1.11.0 works on all of the 4 snes tested. I was able to run through the entire trailer of Akira, Super Road Blaster works. Thank you!

mrehkopf, 2023-03-09:
Ah that's good to know, thanks! I'd like to try different approaches (due to interaction with other issues) so would be great if you could try maybe 1-2 other fpga_base.bit files that I'm going to attach here :)

kimbapslice, 2023-03-09:
Yes, I can do that and report back. Thanks!

mrehkopf, 2023-03-12:
Awesome! Here is another one: 
[fpga_base_new_address_sampling.zip](https://github.com/mrehkopf/sd2snes/files/10951891/fpga_base_new_address_sampling.zip)

kimbapslice, 2023-03-12:
The new address sampling returned to the only snes working was the snes jr systems.

1chip-01 and non-1chip = msu1 video glitches
snes jr stock = works

https://user-images.githubusercontent.com/58018661/224580554-1dd25cd5-4b95-45da-b539-1e1f1d14efdb.mp4

https://user-images.githubusercontent.com/58018661/224580555-5824fce1-d974-4d07-aea6-0ff5a6b03f56.mp4

https://user-images.githubusercontent.com/58018661/224580556-9fb253a7-6fc8-43ac-96c6-714ec939d530.mp4

mrehkopf, 2023-03-13:
Another try 😅 
[fpga_base_new_address_sampling2.zip](https://github.com/mrehkopf/sd2snes/files/10952695/fpga_base_new_address_sampling2.zip)

kimbapslice, 2023-03-13:
Sampling2 results:

non1chip and 1chip-01: glitches, but sometimes shows briefly correctly the first 1-5 seconds of akira/music videos. Super Road Blaster glitches much less but it's still messes up on and off.

snes jr: works perfectly

mrehkopf, 2023-03-13:
Yet another try 🙏
[fpga_base_new_address_sampling3.zip](https://github.com/mrehkopf/sd2snes/files/10955548/fpga_base_new_address_sampling3.zip)
If this doesn't work I'll probably stick with the MSU specific fix for now.

kimbapslice, 2023-03-13:
Sampling3 results:

non1chip and 1chip-01: glitches
snes jr: works perfectly

mrehkopf, 2023-04-10:
OK, went with the first approach for the time being. Thanks for all the testing :) v1.11.1 is around the corner, feel free to re-open the issue if any issues persist.

closed 2023-04-10 by mrehkopf