mrehkopf/sd2snes #162: MSU1 Regression: Resume support appears to not work on 1.11.0b1

Dizzy611, 2021-09-23:
My MSU-1 patch (Dancing Mad) no longer properly resumes audio after battle as of 1.11.0b1 despite working fine in 1.10.3. The first report I had of this was here: https://github.com/Dizzy611/DancingMadFF6/issues/104. I could not initially reproduce this myself due to not owning an SD2SNES/FX Pak, but I recently got one and can definitely confirm this happens.

It's worth noting my code checks the reported MSU-1 revision in order to determine whether or not to enable resume support, however it checks for >2, so unless the revision was decreased this should not have caused issues.

Dizzy611, 2021-09-23:
Confirmed that this does appear to be an issue with the reported revision of the MSU-1. Removing revision checking from my patch seems to resolve the issue.

mrehkopf, 2023-02-05:
I double-checked that MSU-1 returns version 2 in v1.11.0, could you retest with your version check enabled?