mrehkopf/sd2snes #164: 1CHIP Transient Fixes cause Black Screen in Aero Fighters

Dizzy611, 2021-09-24:
Turning on the 1CHIP Transient Fixes on a 2/1/2 SNES with firmware 1.11.0b1 causes a black screen when gameplay begins on Aero Fighters

UltyT, 2021-12-24:
I kinda wonder if this issue is related to the flickering issue I have in FFIII/VI when the game is paused during a battle with the 1CHIP transient fixes option on. I turn it off and the flickering is gone.

DavidOliM, 2023-05-11:
> I kinda wonder if this issue is related to the flickering issue I have in FFIII/VI when the game is paused during a battle with the 1CHIP transient fixes option on. I turn it off and the flickering is gone.

I have a similar graphical issue when a battle fades in on FF5 with the 1chip transient fixes on.